matthewwall/weewx-sdr #152: Add new Sensor Module

db5fp, 2022-03-04:
Hi,

where is the right place to put in a new Sensor Module Bresser5in1 with these datas:

Mar  4 22:56:37 rpi weewx[546] DEBUG user.sdr: lines=[u'{"time" : "2022-03-04 21:56:34", "model" : "Bresser-5in1", "id" : 236, "battery_ok" : 1, "temperature_C" : -2.500, "humidity" : 87, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 180.000, "rain_mm" : 3.600, "mic" : "CHECKSUM"}\n']

Or how do I translate these datas to a vule that is understuud by the software.

Thanks
Frank

db5fp, 2022-03-04:
Update: 
Mar  5 00:16:13 rpi weewx[554] DEBUG user.sdr: lines=[u'{"time" : "2022-03-04 23:16:10", "model" : "Bresser-5in1", "id" : 236, "battery_ok" : 1, "temperature_C" : -3.100, "humidity" : 87, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.200, "wind_dir_deg" : 157.500, "rain_mm" : 3.600, "mic" : "CHECKSUM"}\n']
Mar  5 00:16:13 rpi weewx[554] DEBUG user.sdr: packet={u'outHumidity': 87.0, 'dateTime': 1646435770, u'windDir': 157.5, u'outTemp': -3.1, u'rain_total': 3.6, 'usUnits': 17}

This is the code I've added to sdr.py, but some datas are missing: class BRESSER5in1Packet(Packet):
    # user.sdr: lines=[u'{"time" : "2022-03-04 22:06:10", "model" : "Bresser-5in1", "id" : 236, "battery_ok" : 1, "temperature_C" : -2.400, "humidity" : 86, "wind_max_m_s" : 0.800, "wind_avg_m_s" : 0.800, "wind_dir_deg" : 180.000, "rain$
    IDENTIFIER = "Bresser-5in1"

    @staticmethod
    def parse_json(obj):
        pkt = dict()
        pkt['dateTime'] = Packet.parse_time(obj.get('time'))
        pkt['station_id'] = obj.get('id')
        pkt['battery'] = 0 if obj.get('battery') == 'OK' else 1
        pkt['temperature'] = Packet.get_float(obj, 'temperature_C')
        pkt['humidity'] = Packet.get_float(obj, 'humidity')
        pkt['wind_gust'] = Packet.get_float(obj, 'gust_speed_ms')
        pkt['wind_speed'] = Packet.get_float(obj, 'wind_speed_ms')
        pkt['wind_dir'] = Packet.get_float(obj, 'wind_dir_reg')
        pkt['rain_total'] = Packet.get_float(obj, 'rainfall_mm')
        return Bresser5in1Packet.insert_ids(pkt)

And this is the weewx.con sdr section:

[SDR]

    # This section is for the software-defined radio driver.

    # The time (in seconds) between LOOP packets.
    loop_interval = 2.5

    # Direct testing command:
    # PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/sdr.py --cmd="rtl_433 -M utc -F json -f 914980000 -s 250000"
    # Command probably needs a -p frequency offset parameter.

    # The driver to use
    driver = user.sdr
    path = /usr/share/weewx/user/
    # ld_library_path = /usr/local/include
    cmd = rtl_433 -M utc -F json -R 119 -f 868M
    [[sensor_map]]
        outTemp = temperature.236.Bresser5in1Packet
        outHumidity = humidity.236.Bresser5in1Packet
        windDir = wind_dir.236.Bresser5in1Packet
        windSpeed = wind_speed.236.Bresser5in1Packet
        windGust = wind_gust.236.Bresser5in1Packet
        rain_total = rain_total.236.Bresser5in1Packet
        # uv = uv.16.FOWH65BAltPacket # Not displaying in web page
        # light = light.16.FOWH65BAltPacket # Not displaying in web page
        outTempBatteryStatus = battery.236.Bresser5in1Packet
    [[deltas]]
        rain = rain_total

matthewwall, 2024-12-30:
this has been incorporated into weewx-sdr

closed 2024-12-30 by matthewwall