matthewwall/weewx-sdr #153: WS80 Support

fankyy, 2022-03-05:
The WS80 doesn't seem to be in sdr.py, is there already some work in progress or are there plans to add it?

Because it seems like it was being implemented in rtl_433 recently: https://groups.google.com/g/rtl_433/c/UtzS-OwLLzA/m/p5YCnqxBBAAJ

(Couldn't try it out so far as I'm still waiting for my Ecowitt WS80 to arrive)

andylittle, 2022-03-05:
Run the sdr driver from the command line when your WS80 arrives and post a bit of the output here.

fankyy, 2022-07-05:
My older rtl_433 version on my Raspberry Pi Debian Buster seems to not recognize the WS80, and it seems I have to compile it myself as there are no recent releases for Debian/Raspbian 10, any hint how to start? I've never compiled before

fankyy, 2022-07-05:
Until now I can't get a signal where the gw1100 has 3 out of 4 signal bars. The sdr stick has a nooelec R820T2 chip with antenna set to 17.5cm. I installed the latest firmware from 03/22 (v1.2.1) on the WS80, has something changed or is the signal too weak to be picked up by my sdr stick? The antenna is placed indoors, only around 70 meters away from the sensor, but a building in-between upwards a hill

fankyy, 2022-07-06:
Out of nowhere the packets started showing up in the command line, probably had some loose antenna connection before, anyway, here's some output with `rtl_433 -f 868.3M -M time:utc -M protocol -M level -Y classic -s 300k`

```
time      : 2022-07-06 20:12:56                    Protocol  : 213
model     : Fineoffset-WS80                        ID        : 090026
Battery   : 1.170        Battery Voltage: 3280 mV  Temperature: 19.0 C
Humidity  : 61 %         Wind direction: 304       Wind speed: 0.6 m/s
Gust speed: 0.9 m/s      UVI       : 0.0           Light     : 0.0 lux
Flags     : aa           Integrity : CRC
Modulation: FSK          Freq1     : 868.4 MHz     Freq2     : 868.3 MHz
RSSI      : -12.1 dB     SNR       : 17.4 dB       Noise     : -29.6 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-06 20:13:01                    Protocol  : 213
model     : Fineoffset-WS80                        ID        : 090026
Battery   : 1.170        Battery Voltage: 3280 mV  Temperature: 19.0 C
Humidity  : 61 %         Wind direction: 276       Wind speed: 0.6 m/s
Gust speed: 0.9 m/s      UVI       : 0.0           Light     : 0.0 lux
Flags     : aa           Integrity : CRC
Modulation: FSK          Freq1     : 868.4 MHz     Freq2     : 868.3 MHz
RSSI      : -12.1 dB     SNR       : 18.2 dB       Noise     : -30.4 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-06 20:13:06                    Protocol  : 213
model     : Fineoffset-WS80                        ID        : 090026
Battery   : 1.170        Battery Voltage: 3280 mV  Temperature: 19.0 C
Humidity  : 61 %         Wind direction: 291       Wind speed: 0.0 m/s
Gust speed: 0.8 m/s      UVI       : 0.0           Light     : 0.0 lux
Flags     : aa           Integrity : CRC
Modulation: FSK          Freq1     : 868.4 MHz     Freq2     : 868.3 MHz
RSSI      : -12.1 dB     SNR       : 16.4 dB       Noise     : -28.5 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-06 20:13:11                    Protocol  : 213
model     : Fineoffset-WS80                        ID        : 090026
Battery   : 1.170        Battery Voltage: 3280 mV  Temperature: 19.0 C
Humidity  : 61 %         Wind direction: 313       Wind speed: 0.6 m/s
Gust speed: 0.9 m/s      UVI       : 0.0           Light     : 0.0 lux
Flags     : aa           Integrity : CRC
Modulation: FSK          Freq1     : 868.4 MHz     Freq2     : 868.3 MHz
RSSI      : -12.1 dB     SNR       : 16.0 dB       Noise     : -28.2 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-06 20:13:15                    Protocol  : 213
model     : Fineoffset-WS80                        ID        : 090026
Battery   : 1.170        Battery Voltage: 3280 mV  Temperature: 19.0 C
Humidity  : 61 %         Wind direction: 323       Wind speed: 0.7 m/s
Gust speed: 0.9 m/s      UVI       : 0.0           Light     : 0.0 lux
Flags     : aa           Integrity : CRC
Modulation: FSK          Freq1     : 868.4 MHz     Freq2     : 868.3 MHz
RSSI      : -12.1 dB     SNR       : 17.4 dB       Noise     : -29.6 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-06 20:13:20                    Protocol  : 213
model     : Fineoffset-WS80                        ID        : 090026
Battery   : 1.170        Battery Voltage: 3280 mV  Temperature: 19.0 C
Humidity  : 61 %         Wind direction: 336       Wind speed: 0.7 m/s
Gust speed: 1.0 m/s      UVI       : 0.0           Light     : 0.0 lux
Flags     : aa           Integrity : CRC
Modulation: FSK          Freq1     : 868.4 MHz     Freq2     : 868.3 MHz
RSSI      : -12.1 dB     SNR       : 19.2 dB       Noise     : -31.4 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
```

Do I have to post output with other parameters or is this enough for the sensor map entries?

andylittle, 2022-07-06:
get me the json output please rtl_433 -M utc -F json

fankyy, 2022-07-06:
`{"time" : "2022-07-06 21:00:31", "protocol" : 213, "model" : "Fineoffset-WS80", "id" : 589862, "battery_ok" : 1.170, "battery_mV" : 3280, "temperature_C" : 17.900, "humidity" : 66, "wind_dir_deg" : 313, "wind_avg_m_s" : 2.000, "wind_max_m_s" : 2.000, "uvi" : 0.000, "light_lux" : 0.000, "flags" : 170, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.385, "freq2" : 868.316, "rssi" : -12.144, "snr" : 15.850, "noise" : -27.994}
{"time" : "2022-07-06 21:00:36", "protocol" : 213, "model" : "Fineoffset-WS80", "id" : 589862, "battery_ok" : 1.170, "battery_mV" : 3280, "temperature_C" : 17.900, "humidity" : 66, "wind_dir_deg" : 310, "wind_avg_m_s" : 1.600, "wind_max_m_s" : 2.000, "uvi" : 0.000, "light_lux" : 0.000, "flags" : 170, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.386, "freq2" : 868.303, "rssi" : -12.144, "snr" : 18.239, "noise" : -30.383}
{"time" : "2022-07-06 21:00:41", "protocol" : 213, "model" : "Fineoffset-WS80", "id" : 589862, "battery_ok" : 1.170, "battery_mV" : 3280, "temperature_C" : 17.900, "humidity" : 66, "wind_dir_deg" : 300, "wind_avg_m_s" : 1.600, "wind_max_m_s" : 2.000, "uvi" : 0.000, "light_lux" : 0.000, "flags" : 170, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.385, "freq2" : 868.318, "rssi" : -12.144, "snr" : 18.539, "noise" : -30.683}`

andylittle, 2022-07-06:
replace your sdr.py driver with [this](https://github.com/andylittle/weewx-sdr-1/blob/master/bin/user/sdr.py) one. Let me know.

Not sure what "battery_ok" : 1.170 this indicates, ususally is a 1 or ) based on battery status.

fankyy, 2022-07-06:
Thanks a ton! I just tried with this sensor map:

```
	outTemp = temperature.*.FOWS80Packet
        windSpeed = wind_speed.*.FOWS80Packet
        windDir = wind_dir.*.FOWS80Packet
        windGust = wind_gust.*.FOWS80Packet
        outHumidity = humidity.*.FOWS80Packet
        rain_total = rain_total.*.FOWS80Packet
        UV = uv_index.*.FOWS80Packet
        radiation = light.*.FOWS80Packet
	illuminance = light.*.FOWS80Packet
	luminosity = light.*.FOWS80Packet
# txBatteryStatus = battery.*.FOWS80Packet
```
But there seems to be no report until now. (nothing after 'starting main packet loop'). Do I have to specify the ID? (which one by the way, the one from the json output seems to differ?) Could the voltage be included somehow in the future?

matthewwall, 2024-12-30:
the FOWS80Packet was merged in with other changes from andylittle.  thank you andy!

closed 2024-12-30 by matthewwall