matthewwall/weewx-sdr #157: Parser for Kedsum temperature humidity sensor

Sar6e, 2022-06-22:
The USB console for my WH1080 station failed some time ago.  I was really pleased to be able to continue to use the sensor array for the station using weewx-sdr.  Thanks so much for your work!  The instructions were clear and simple to follow.  Great work!

Now that I have no console I've lost inside temperature and humidity.  I have a temperature and humidity sensor that came with a set of kitchen scales of all things.  I can see this output when running manually.

out: ['{"time" : "2022-06-17 00:23:59", "model" : "Kedsum-TH", "id" : 235, "channel" : 1, "battery_ok" : 0, "flags" : 8, "temperature_F" : 59.000, "humidity" : 74, "mic" : "CRC"}\n']

I'm not sure if creating a parser is within my capabilities or not but haven't been able to identify any pointers on how I might go about it.

Sar6e, 2022-06-23:
I renamed the existing sdr.py file then downloaded the updated sdr.py file.  When I try to run the driver manually, I get an error.

  File "/usr/share/weewx/user/sdr.py", line 8
    <!DOCTYPE html>
    ^
SyntaxError: invalid syntax

Should I have done something more than just update the sdr.py file?

andylittle, 2022-06-23:
How did you run it? If you installed in /home/weewx
`cd /home/weewx
sudo PYTHONPATH=bin python bin/user/sdr.py --cmd="rtl_433 -M utc -F json"`

Sar6e, 2022-06-23:
I originally installed using wee-extension.  In /usr/share/weewx/user I now have sdr-dist.py which is a copy of the original sdr.py.  I also have sdr-kedsum.py which is a download of your latest commit.  

If I copy sdr-dist.py to sdr.py and execute 
`sudo PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/sdr.py --cmd="rtl_433 -M utc -F json"`
I start to see data being captured and parsed.

If I copy sdr-kedsum.py to sdr.py and execute the same command I get the syntax error.

Sar6e, 2022-06-23:
To answer your other question...  I have no idea what the flag attribute is returning.

andylittle, 2022-06-23:
https://github.com/andylittle/weewx-sdr-1/blob/16c92a043450db06f2e662146a08b76801522230/bin/user/sdr.py maybe that link was wrong.

Sar6e, 2022-06-23:
Hi,

I just grabbed downloaded from the new link provided.  Unfortunately, when I use that version of sdr.py, I'm still getting the syntax error.

Sar6e, 2022-06-25:
I grabbed the additions you made to sdr.py and added them to my original copy of sdr.py.  The driver is running manually and parsing the Kedsum data successfully.  After updating weewx.conf, the values are writing to the DB successfully.  

Happy to continue to test additional updates if that helps.

matthewwall, 2024-12-30:
this has been incorporated into weewx-sdr

closed 2024-12-30 by matthewwall