matthewwall/weewx-sdr #159: Fine Offset WH1080 not parsing battery status correctly

Sar6e, 2022-07-06:
The output from the sensor is

out: ['{"time" : "2022-07-06 11:36:54", "model" : "Fineoffset-WHx080", "subtype" : 0, "id" : 107, "battery_ok" : 1, "temperature_C" : 11.000, "humidity" : 73, "wind_dir_deg" : 45, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "rain_mm" : 1223.400, "mic" : "CRC"}\n']

The raw output of  "battery_ok":1 is being parsed to  'battery.107.FOWHx080Packet': 0

parsed: {'dateTime': 1657107414, 'usUnits': 16, 'msg_type.107.FOWHx080Packet': 0, 'temperature.107.FOWHx080Packet': 11.0, 'humidity.107.FOWHx080Packet': 73.0, 'wind_dir.107.FOWHx080Packet': 45.0, 'wind_speed.107.FOWHx080Packet': 0.0, 'wind_gust.107.FOWHx080Packet': 0.0, 'rain_total.107.FOWHx080Packet': 122.34, 'battery.107.FOWHx080Packet': 0, 'signal_type.107.FOWHx080Packet': 0, 'hours.107.FOWHx080Packet': None, 'minutes.107.FOWHx080Packet': None, 'seconds.107.FOWHx080Packet': None, 'year.107.FOWHx080Packet': None, 'month.107.FOWHx080Packet': None, 'day.107.FOWHx080Packet': None}

andylittle, 2022-07-07:
This comment is from the sdr.py driver, does this help? Or am I missing something?

`Battery Status

In the weewx database, a battery status of 1 indicates low battery.  This has
origins in the original battery indicators from davis vantage stations.  Some
devices report 'battery' where a value of 1 indicates that the battery is ok,
i.e., the battery is *not* low.  `

Sar6e, 2022-07-12:
OK, that's interesting to understand.  

The WH1080 sensor array is reporting in the raw data "battery_ok" : 1, which is being parsed to 'battery.107.FOWHx080Packet': 0.
The Kedsum-TH is reporting in the raw data "battery_ok" : 0, which is being parsed to 'battery.179.KedsumTHPacket': 1.

I should probably change the batteries in both sensors to understand what value they report when the battery is fully charged.

matthewwall, 2024-12-30:
the boolean battery indicator has been consolidated, so the different rtl_433 outputs of 'battery' or 'battery_ok' should be handled correctly now.  however there is still some uncertainty with some of the recent (2024) ecowitt sensors - some of them emit a floating value for 'battery_ok', and some emit the battery voltage as well.

closed 2024-12-30 by matthewwall