matthewwall/weewx-sdr #161: Rain totals not right

whtz, 2022-07-27:
I'm new to all of this, so I'm sure it's probably something simple. 

I'm running weewx and weewx-sdr with a sainlogic professional all in one.  The sainlogic shows .2 inches of rain over the past 24 hours, weewx says 2.01 inches.  I don't believe I changed anything to do with calculating, but like I said, I'm brand new to all of this. I am kinda wondering about the line in my [[sensor_map]]:
  [[deltas]] rain=rain_total

Thanks for any help!

"Details" of my installation:
I have the sainlogic "professional" weather station WS0835 (who knew the "professional" model didn't keep stats?)  I wanted history, so started digging and found weewx and sdr.

I believe I have installed weewx, weewx-sdr, and made changes similar to this thread (https://github.com/matthewwall/weewx-sdr/issues/148) following two main sources as my guide (https://sh.com.hr/hacking-the-ft-0310-weather-station/, https://www.oneeyedman.net/posts/2020052301-weatherstation.html)

  [[sensor_map]]
        windDir = wind_dir.31.Cotech367959Packet
        windSpeed = wind_speed.31.Cotech367959Packet
        windGust = wind_gust.31.Cotech367959Packet
        outTemp = temperature.31.Cotech367959Packet
        outHumidity = humidity.31.Cotech367959Packet
        rain_total = total_rain.31.Cotech367959Packet
        UV = uv.31.Cotech367959Packet
        luminosity = light_lux.31.Cotech367959Packet
    [[deltas]]
        rain = rain_total

andylittle, 2022-07-28:
Your delta looks correct. From the comments in the sdr.py weewx driver:

`
The deltas stanza indicates which observations are cumulative measures and
how they should be split into delta measures.

[SDR]
    ...
    [[deltas]]
        rain = rain_total

In this case, the value for rain will be a delta calculated from sequential
rain_total observations.
`
Set debug = 1 at the start of your weewx.conf file and restart weewx. Look at the messages coming from your station.

    #{"time" : "2022-03-01 14:11:42", "model" : "Cotech-367959", "id" : 24, "battery_ok" : 1, "temperature_F" : 46.900, "humidity" : 62, "rain_mm" : 18.600, "wind_dir_deg" : 16, "wind_avg_m_s" : 0.600, "wind_max_m_s" : 0.700, "mic" : "CRC"}
   
Is the station really sending rain in mm?

You will get more eyes on the issue over [here](https://groups.google.com/g/weewx-user).

Andy

whtz, 2022-07-28:
Thanks. I've done the debug thing but my messages don't look like that:

`ul 28 14:33:50 Pgriffy-LinuxMint weewx[59608] DEBUG user.sdr: packet={'windDir': 94.0, 'windSpeed': 0.5, 'windGust': 0.7, 'outTemp': 31.555555555555557, 'outHumidity': 48.0, 'rain_total': 66.6, 'UV': 25.1, 'luminosity': 69627, 'dateTime': 1659033227, 'usUnits': 16}
`
I'll make a new post over at the other place if the above isn't an obvious, "oh yeah, just flip this switch on" type fix.

andylittle, 2022-07-28:
That message shows what the driver parsed.  We need an example of what the device sent.

closed 2024-12-30 by matthewwall