matthewwall/weewx-sdr #162: new insertion in sensor map

farmabep, 2022-08-10:
Hello everyone, I ask you for a little help to insert the uv and radiation sensors of a whx080 station into my sensor map.
Here is the result of the sudo rtl_433 -M utc -F json command:

{"time": "2022-08-07 13:26:39", "model": "Fineoffset-WHx080", "subtype": 2, "uv_sensor_id": 225, "uv_status": "OK", "uv_index ": 2," lux ": 32082.000," wm ": 46,972," mic ":" CRC "}

What is the right line to insert within the weewx.conf sensor map?
thanks for your cooperation and sorry for my bad english
I tried with:
UV = uv_index.225.FOWHx080Packet
radiation = light.225.FOWHx080Packet
radiation = radiation.225.FOWHx080Packet
UV = uv_index.225.FOWH3080Packet
radiation = light.225.FOWH3080Packet
radiation = lradiation.225.FOWH3080Packet
I don't know what I did wrong.

i used the command: PYTHONPATH=/home/weewx/bin python /home/weewx/bin/user/sdr.py --cmd "rtl_433 -M utc" --hide parsed,empty     
and this is the result:                    _ _ _ \n', 'time      : 2022-08-10 16:02:35\n', 'model     : Fineoffset-WHx080                      Msg type  : 2             UV Sensor ID: 225         Sensor Status: OK         UV Index  : 2             Lux       : 49540.2       Watts/m   : 72.53         Integrity : CRC\n', '_ _ _ _ thanks for your cooperation and your help

farmabep, 2022-08-14:
sudo PYTHONPATH=/home/weewx/bin python /home/weewx/bin/user/sdr.py --hide parsed, mapped, empty

out: ['{"time" : "2022-08-14 17:22:30", "model" : "Fineoffset-WHx080", "subtype" : 2, "uv_sensor_id" : 225, "uv_status" : "OK", "uv_index" : 1, "lux" : 2223.200, "wm" : 3.255, "mic" : "CRC"}\n', '{"time" : "2022-08-14 17:22:30", "model" : "Fineoffset-WHx080", "subtype" : 2, "uv_sensor_id" : 225, "uv_status" : "OK", "uv_index" : 1, "lux" : 2223.200, "wm" : 3.255, "mic" : "CRC"}\n', '{"time" : "2022-08-14 17:22:30", "model" : "Fineoffset-WHx080", "subtype" : 2, "uv_sensor_id" : 225, "uv_status" : "OK", "uv_index" : 1, "lux" : 2223.200, "wm" : 3.255, "mic" : "CRC"}\n']

fankyy, 2022-08-17:
This is not a sensor map issue, it's an identifier issue. There are a lot of WH2310/WH3080 models identifying with `Fineoffset-WHx080`, but sdr.py only has uv/radiation parsing for these in

`class FOWH3080Packet(Packet):`

with    ` IDENTIFIER = "Fine Offset Electronics WH3080 Weather Station"`

Not sure if an identifier change (adding  `IDENTIFIER = "Fineoffset-WHx080"` to this class) does work or if there needs more code to be changed?

farmabep, 2022-08-17:
First of all, thank you for your reply
it's really true .
This IDENTIFIER = "Fine Offset Electronics WH3080 Weather Station" could therefore refer to the newer station type wh3080 with sensors all in one.
in fact my station is the froggit wh3080 with separate sensors.
It is very difficult for me to touch or modify the sdr.py
in fact my identifier is Fine Offset Electronics WHx080 Weather Station "whose package is FOWHx080Packet and not FOWH3080Packet.
I hope that someone of good will will be able to help me in inserting the right lines in the sdr.py to make sure that weewx also recognizes the two UV and Radiation sensors

andylittle, 2022-08-17:
Does your weather station send other messages besides this one? I assume it does.  Please post a sample of all the different type messages it sends.
``` {"time" : "2022-08-14 17:22:30", "model" : "Fineoffset-WHx080", "subtype" : 2, "uv_sensor_id" : 225, "uv_status" : "OK", "uv_index" : 1, "lux" : 2223.200, "wm" : 3.255, "mic" : "CRC"}```

andylittle, 2022-08-17:
@fankyy  If the stations that send "Fineoffset-WHx080" as the model do not all send the same packets, we could do something like the Acurite5n1 which sends different packets.

farmabep, 2022-08-17:
good evening Andylittle
here is the rtl_433 command:
time: 2022-08-17 17:42:26
model: Fineoffset-WHx080 Msg type: 2
UV Sensor ID: 225 Sensor Status: OK UV Index: 2
Lux: 53419.8 Watts / m: 78.21 Integrity: CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time: 2022-08-17 17:42:26
model: Fineoffset-WHx080 Msg type: 2
UV Sensor ID: 225 Sensor Status: OK UV Index: 2
Lux: 53419.8 Watts / m: 78.21 Integrity: CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time: 2022-08-17 17:42:26
model: Fineoffset-WHx080 Msg type: 2
UV Sensor ID: 225 Sensor Status: OK UV Index: 2
Lux: 53419.8 Watts / m: 78.21 Integrity: CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time: 2022-08-17 17:42:27
model: Acurite-606TX id: 92
Battery: 0 Temperature: 34.0 C Integrity: CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time: 2022-08-17 17:42:30
model: Acurite-6045M id: 193
channel: C Battery: 1 temperature: 96.1 F
humidity: 36% strike_count: 217 storm_distance: 27
active_mode: 0 rfi_detect: 0 data_exception: 0
raw_message: 00c16f2493096cdb37
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time: 2022-08-17 17:42:30
model: Acurite-6045M id: 193
channel: C Battery: 1 temperature: 96.1 F
humidity: 36% strike_count: 217 storm_distance: 27
active_mode: 0 rfi_detect: 0 data_exception: 0
raw_message: 00c16f2493096cdb37
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time: 2022-08-17 17:42:30
model: Acurite-6045M id: 193
channel: C Battery: 1 temperature: 96.1 F
humidity: 36% strike_count: 217 storm_distance: 27
active_mode: 0 rfi_detect: 0 data_exception: 0
raw_message: 00c16f2493096cdb37
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time: 2022-08-17 17:42:39
model: Fineoffset-WHx080 Msg type: 0
Station ID: 226
Battery: 1 Temperature: 35.6 C Humidity: 28%
Wind Direction: 0 Wind avg speed: 0.00 Wind gust: 0.00
Total rainfall: 65.1 Integrity: CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time: 2022-08-17 17:42:47
model: Hideki-TS04 Rolling Code: 12
Channel: 3 Battery: 1 Temperature: 34.3 C
Humidity: 43% Integrity: CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time: 2022-08-17 17:42:47
model: Hideki-TS04 Rolling Code: 12
Channel: 3 Battery: 1 Temperature: 34.3 C
Humidity: 43% Integrity: CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time: 2022-08-17 17:42:47
model: Hideki-TS04 Rolling Code: 12
Channel: 3 Battery: 1 Temperature: 34.3 C
Humidity: 43% Integrity: CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time: 2022-08-17 17:42:50
model: Hideki-Wind Rolling Code: 8
Channel: 4 Battery: 1 Temperature: 37.6 C
Wind Speed: 2.30 mph Gust Speed: 1.50 mph Wind Approach: 1
Wind Direction: 180.0 Integrity: CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time: 2022-08-17 17:42:50
model: Hideki-Wind Rolling Code: 8
Channel: 4 Battery: 1 Temperature: 37.6 C
Wind Speed: 2.30 mph Gust Speed: 1.50 mph Wind Approach: 1
Wind Direction: 180.0 Integrity: CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time: 2022-08-17 17:42:50
model: Hideki-Wind Rolling Code: 8
Channel: 4 Battery: 1 Temperature: 37.6 C
Wind Speed: 2.30 mph Gust Speed: 1.50 mph Wind Approach: 1
Wind Direction: 180.0 Integrity: CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time: 2022-08-17 17:42:54
model: Acurite-6045M id: 193
channel: C Battery: 1 temperature: 96.1 F
humidity: 36% strike_count: 217 storm_distance: 27
active_mode: 0 rfi_detect: 0 data_exception: 0
raw_message: 00c16f2493096cdb37
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
thanks for coming

andylittle, 2022-08-17:
```rtl_433 -M utc -F json```   and just the message from the  Fineoffset-WHx080

andylittle, 2022-08-17:
@farmabep   try [this](https://github.com/andylittle/weewx-sdr-1/commit/ff72f55cf5ca90d645de6a47a5d7b2a641785e07) one

fankyy, 2022-08-18:
> @fankyy If the stations that send "Fineoffset-WHx080" as the model do not all send the same packets, we could do something like the Acurite5n1 which sends different packets.

I think, regarding uv/radiation they all behave the same, depending on whether or not it is available and maybe a new enough transmitter. AFAIK, the uv/solar packets are always sent with their own ID, as a separate packet, but using the same transmitter.

But more interesting would be, why the wh2315 (wh2310) display console seems to get the barometric pressure data from the transmitter. If I find some time for it, i'll record the raw data

farmabep, 2022-08-18:
I'm sorry Andy
but with your changes the sdr.py doesn't work
I tried to insert in the sensor map:
UV = uv_index.225.FOWH3080Packet
and also as a replacement: UV = uv_index.225.FOWHx080Packet
A small note: at the suggestion of fankyy I tried to replace IDENTIFIER = "Fine Offset Electronics WH3080 Weather Station"
with
  IDENTIFIER = "Endoffset-WHx080"
weewx incredibly reads uv and radiation data but no longer reads temp, rain and humidity data
could it be the right way?
Ulteriori informazioni su questo testo di originePer avere ulteriori informazioni sulla traduzione è necessario il testo di origine
Invia commenti
Riquadri laterali

farmabep, 2022-08-18:
I could attach if you want the weewx.conf and how the sdr.py works and a large portion of syslog

andylittle, 2022-08-18:
Use this in your sensor map. FOWHx080Packet 

Run sdr.py from the command line and attach the output. Also post your sensor map.

farmabep, 2022-08-18:
PYTHONPATH=/home/weewx/bin python /home/weewx/bin/user/sdr.py --config /home/weewx/weewx.conf
out: ['{"time" : "2022-08-18 19:07:34", "model" : "Acurite-606TX", "id" : 92, "battery_ok" : 0, "temperature_C" : 28.100, "mic" : "CHECKSUM"}\n']
parsed: {'dateTime': 1660849654, 'usUnits': 1, 'temperature.92.Acurite606TXPacketV2': 82.58000000000001, 'battery.92.Acurite606TXPacketV2': 1}
unmapped: {'dateTime': 1660849654, 'usUnits': 1, 'temperature.92.Acurite606TXPacketV2': 82.58000000000001, 'battery.92.Acurite606TXPacketV2': 1}
out: ['{"time" : "2022-08-18 19:07:47", "model" : "Hideki-TS04", "id" : 12, "channel" : 3, "battery_ok" : 1, "temperature_C" : 29.400, "humidity" : 57, "mic" : "CRC"}\n', '{"time" : "2022-08-18 19:07:48", "model" : "Hideki-TS04", "id" : 12, "channel" : 3, "battery_ok" : 1, "temperature_C" : 29.400, "humidity" : 57, "mic" : "CRC"}\n', '{"time" : "2022-08-18 19:07:48", "model" : "Acurite-6045M", "id" : 193, "channel" : "C", "battery_ok" : 1, "temperature_F" : 82.200, "humidity" : 57, "strike_count" : 253, "storm_dist" : 22, "active" : 0, "rfi" : 0, "exception" : 0, "raw_msg" : "00c16f39117e7e56cc"}\n']
parsed: {'dateTime': 1660849667, 'usUnits': 16, 'temperature.3:12.HidekiTS04Packet': 29.4, 'humidity.3:12.HidekiTS04Packet': 57.0, 'battery.3:12.HidekiTS04Packet': 0}
mapped: {'inTemp': 29.4, 'inTempBatteryStatus': 0, 'dateTime': 1660849667, 'usUnits': 16}
parsed: {'dateTime': 1660849668, 'usUnits': 16, 'temperature.3:12.HidekiTS04Packet': 29.4, 'humidity.3:12.HidekiTS04Packet': 57.0, 'battery.3:12.HidekiTS04Packet': 0}
mapped: {'inTemp': 29.4, 'inTempBatteryStatus': 0, 'dateTime': 1660849668, 'usUnits': 16}
parsed: {'dateTime': 1660849668, 'usUnits': 1, 'channel.00C1.AcuriteLightningPacket': 'C', 'temperature.00C1.AcuriteLightningPacket': 82.2, 'battery.00C1.AcuriteLightningPacket': 0, 'humidity.00C1.AcuriteLightningPacket': 57, 'active.00C1.AcuriteLightningPacket': 0, 'rfi.00C1.AcuriteLightningPacket': 0, 'exception.00C1.AcuriteLightningPacket': 0, 'strikes_total.00C1.AcuriteLightningPacket': 253, 'distance.00C1.AcuriteLightningPacket': 22}
mapped: {'extraTemp1': 82.2, 'active': 0, 'distance': 22, 'rfi': 0, 'battery': 0, 'lightning_strikes_total': 253, 'dateTime': 1660849668, 'usUnits': 1}
out: ['{"time" : "2022-08-18 19:07:52", "model" : "Hideki-Wind", "id" : 8, "channel" : 4, "battery_ok" : 1, "temperature_C" : 26.700, "wind_avg_mi_h" : 2.400, "wind_max_mi_h" : 4.900, "wind_approach" : 1, "wind_dir_deg" : 225.000, "mic" : "CRC"}\n', '{"time" : "2022-08-18 19:07:52", "model" : "Hideki-Wind", "id" : 8, "channel" : 4, "battery_ok" : 1, "temperature_C" : 26.700, "wind_avg_mi_h" : 2.400, "wind_max_mi_h" : 4.900, "wind_approach" : 1, "wind_dir_deg" : 225.000, "mic" : "CRC"}\n', '{"time" : "2022-08-18 19:07:53", "model" : "Hideki-Wind", "id" : 8, "channel" : 4, "battery_ok" : 1, "temperature_C" : 26.700, "wind_avg_mi_h" : 2.400, "wind_max_mi_h" : 4.900, "wind_approach" : 1, "wind_dir_deg" : 225.000, "mic" : "CRC"}\n']
parsed: {'dateTime': 1660849672, 'usUnits': 16, 'temperature.4:None.HidekiWindPacket': 26.7, 'wind_speed.4:None.HidekiWindPacket': 3.862416, 'wind_dir.4:None.HidekiWindPacket': 225.0, 'wind_gust.4:None.HidekiWindPacket': 7.885766, 'battery.4:None.HidekiWindPacket': 0}
mapped: {'windDir': 225.0, 'windSpeed': 3.862416, 'windGust': 7.885766, 'windBatteryStatus': 0, 'dateTime': 1660849672, 'usUnits': 16}
parsed: {'dateTime': 1660849672, 'usUnits': 16, 'temperature.4:None.HidekiWindPacket': 26.7, 'wind_speed.4:None.HidekiWindPacket': 3.862416, 'wind_dir.4:None.HidekiWindPacket': 225.0, 'wind_gust.4:None.HidekiWindPacket': 7.885766, 'battery.4:None.HidekiWindPacket': 0}
mapped: {'windDir': 225.0, 'windSpeed': 3.862416, 'windGust': 7.885766, 'windBatteryStatus': 0, 'dateTime': 1660849672, 'usUnits': 16}
parsed: {'dateTime': 1660849673, 'usUnits': 16, 'temperature.4:None.HidekiWindPacket': 26.7, 'wind_speed.4:None.HidekiWindPacket': 3.862416, 'wind_dir.4:None.HidekiWindPacket': 225.0, 'wind_gust.4:None.HidekiWindPacket': 7.885766, 'battery.4:None.HidekiWindPacket': 0}
mapped: {'windDir': 225.0, 'windSpeed': 3.862416, 'windGust': 7.885766, 'windBatteryStatus': 0, 'dateTime': 1660849673, 'usUnits': 16}
out: ['{"time" : "2022-08-18 19:08:05", "model" : "Acurite-606TX", "id" : 92, "battery_ok" : 0, "temperature_C" : 28.100, "mic" : "CHECKSUM"}\n']
parsed: {'dateTime': 1660849685, 'usUnits': 1, 'temperature.92.Acurite606TXPacketV2': 82.58000000000001, 'battery.92.Acurite606TXPacketV2': 1}
unmapped: {'dateTime': 1660849685, 'usUnits': 1, 'temperature.92.Acurite606TXPacketV2': 82.58000000000001, 'battery.92.Acurite606TXPacketV2': 1}
out: ['{"time" : "2022-08-18 19:08:12", "model" : "Acurite-6045M", "id" : 193, "channel" : "C", "battery_ok" : 1, "temperature_F" : 82.200, "humidity" : 57, "strike_count" : 253, "storm_dist" : 22, "active" : 0, "rfi" : 0, "exception" : 0, "raw_msg" : "00c16f39117e7e56cc"}\n', '{"time" : "2022-08-18 19:08:12", "model" : "Acurite-6045M", "id" : 193, "channel" : "C", "battery_ok" : 1, "temperature_F" : 82.200, "humidity" : 57, "strike_count" : 253, "storm_dist" : 22, "active" : 0, "rfi" : 0, "exception" : 0, "raw_msg" : "00c16f39117e7e56cc"}\n', '{"time" : "2022-08-18 19:08:12", "model" : "Acurite-6045M", "id" : 193, "channel" : "C", "battery_ok" : 1, "temperature_F" : 82.200, "humidity" : 57, "strike_count" : 253, "storm_dist" : 22, "active" : 0, "rfi" : 0, "exception" : 0, "raw_msg" : "00c16f39117e7e56cc"}\n', '{"time" : "2022-08-18 19:08:15", "model" : "Fine [... cut ...] 5.100, "mic" : "CRC"}\n']
parsed: {'dateTime': 1660849743, 'usUnits': 16, 'msg_type.226.FOWHx080Packet': 0, 'battery.226.FOWHx080Packet': 0, 'temperature.226.FOWHx080Packet': 27.3, 'humidity.226.FOWHx080Packet': 54.0, 'wind_dir.226.FOWHx080Packet': 0.0, 'wind_speed.226.FOWHx080Packet': 0.0, 'wind_gust.226.FOWHx080Packet': 0.0, 'rain_total.226.FOWHx080Packet': 6.51}
mapped: {'outTemp': 27.3, 'outTempBatteryStatus': 0, 'outHumidity': 54.0, 'rain_total': 6.51, 'dateTime': 1660849743, 'usUnits': 16}

farmabep, 2022-08-18:
[SDR]
    # This section is for the software-defined radio driver.
    
    # The driver to use
    driver = user.sdr
   #  log_unknown_sensors = True
  #  log_unmapped_sensors = True
    cmd = rtl_433 -M utc -F json -R 40 -R 32 -R 42
   #  cmd = rtl_433 -M utc -F json -M oldmodel
    [[sensor_map]]
        outTemp = temperature.226.FOWHx080Packet
        outTempBatteryStatus = battery.226.FOWHx080Packet
        outHumidity = humidity.226.FOWHx080Packet
       # extraTemp2 = temperature.226.FOWHx080Packet
        # uvBatteryStatus = battery.1:184.OSUV800Packet
        UV = uv_index.225.FOWHx080Packet
        radiation = radiation.225.FOWx080Packet
        inTemp = temperature.3:12.HidekiTS04Packet
        inTempBatteryStatus = battery.3:12.HidekiTS04Packet
        extraTemp1 = temperature.00C1.AcuriteLightningPacket
       # outTempBatteryStatus = battery.5:11.HidekiTS04Packet
       # outHumidity = humidity.5:11.HidekiTS04Packet
       # outTemp = temperature.5:11.HidekiTS04Packet
        windDir = wind_dir.4:None.HidekiWindPacket
        windSpeed = wind_speed.4:None.HidekiWindPacket
        windGust = wind_gust.4:None.HidekiWindPacket
        windBatteryStatus = battery.4:None.HidekiWindPacket
       # rainBatteryStatus = battery.4:None.HidekiRainPacket
       # rain_total = rain_total.4:None.HidekiRainPacket
       # rainBatteryStatus = battery.226.FOWHx080Packet
        rain_total = rain_total.226.FOWHx080Packet
        active = active.00C1.AcuriteLightningPacket
        distance = distance.00C1.AcuriteLightningPacket
        rfi = rfi.00C1.AcuriteLightningPacket
        battery = battery.00C1.AcuriteLightningPacket
        lightning_strikes_total = strikes_total.00C1.AcuriteLightningPacket

farmabep, 2022-08-18:
ignore log_unknown_sensors = True
log_unmapped_sensors = True and of the cmd = rtl_433 -M utc -F json -M oldmodel; they have the gate and are deactivated

andylittle, 2022-08-18:
Change line 1356 from if msg_type == 20: to if msg_type == 2:. Or pull the latest from my repo.

farmabep, 2022-08-19:
parsed: {'dateTime': 1660923203, 'usUnits': 16, 'msg_type.225.FOWHx080Packet': 2, 'uv_status.225.FOWHx080Packet': 0, 'uv_index.225.FOWHx080Packet': 3.0, 'luminosity.225.FOWHx080Packet': 56756.5, 'radiation.225.FOWHx080Packet': 83.099}
mapped: {'UV': 3.0, 'dateTime': 1660923203, 'usUnits': 16}
parsed: {'dateTime': 1660923203, 'usUnits': 16, 'msg_type.225.FOWHx080Packet': 2, 'uv_status.225.FOWHx080Packet': 0, 'uv_index.225.FOWHx080Packet': 3.0, 'luminosity.225.FOWHx080Packet': 56756.5, 'radiation.225.FOWHx080Packet': 83.099}
mapped: {'UV': 3.0, 'dateTime': 1660923203, 'usUnits': 16}
parsed: {'dateTime': 1660923203, 'usUnits': 16, 'msg_type.225.FOWHx080Packet': 2, 'uv_status.225.FOWHx080Packet': 0, 'uv_index.225.FOWHx080Packet': 3.0, 'luminosity.225.FOWHx080Packet': 56756.5, 'radiation.225.FOWHx080Packet': 83.099}
mapped: {'UV': 3.0, 'dateTime': 1660923203, 'usUnits': 16}

farmabep, 2022-08-19:
now it works very well ... every now and then we see a new ET evapotranspiration function with a value to 6 7 decimal places. never seen before.
anyway a lot of thanks and I hope to become as good as you are
can we please wait a few days to close this issue?

closed 2022-08-23 by farmabep