matthewwall/weewx-sdr #163: Fineoffset-WH32

mfraser, 2022-08-19:
I have just purchased a WH32 outdoor sensor. It responds as:
{"time" : "2022-08-19 21:34:58", "model" : "Fineoffset-WH32", "id" : 1 3,
"battery_ok" : 1, "temperature_C" : 26.200, "humidity" : 55, "mic" : "CRC"
}

Is this the same as the WH32B?

sandeen, 2024-05-27:
I have one too, and I think the only difference is the lack of the "B" on the Model string .I just cloned the existing WH32B class and edited out the B in sdr.py, but you could probably just as easily edit class FOWH32BPacket(Packet): to remove the "B" everywhere. I should send a pull request ... ;)

```
--- /usr/share/weewx/user/sdr.py	2023-08-23 12:51:27.106343593 -0500
+++ /etc/weewx/bin/user/sdr.py	2024-05-26 17:43:50.048720047 -0500
@@ -1604,6 +1604,41 @@
         return Packet.add_identifiers(pkt, station_id, FOWH32BPacket.__name__)
 
 
+class FOWH32Packet(Packet):
+    # This is for a WH32 which is the indoors sensor array for an Ambient
+    # Weather WS-2902A. The same sensor array is used for several models.
+
+    # time      : 2019-04-08 00:48:02
+    # model     : Fineoffset-WH32
+    # ID        : 146
+    # Temperature: 17.5 C
+    # Humidity  : 60 %
+    # Pressure  : 1001.2 hPa
+    # Battery   : OK
+    # Integrity : CHECKSUM
+
+    # {"time" : "2024-05-26 20:23:03", "model" : "Fineoffset-WH32", "id" : 172, "battery_ok" : 1, "temperature_C" : 21.900, "humidity" : 60, "mic" : "CRC"}
+
+
+    IDENTIFIER = "Fineoffset-WH32"
+
+    @staticmethod
+    def parse_json(obj):
+        pkt = dict()
+        pkt['dateTime'] = Packet.parse_time(obj.get('time'))
+        pkt['usUnits'] = weewx.METRIC
+        pkt['station_id'] = obj.get('id')
+        pkt['temperature'] = Packet.get_float(obj, 'temperature_C')
+        pkt['humidity'] = Packet.get_float(obj, 'humidity')
+        pkt['battery'] = 0 if obj.get('battery') == 'OK' else 1
+        return FOWH32Packet.insert_ids(pkt)
+
+    @staticmethod
+    def insert_ids(pkt):
+        station_id = pkt.pop('station_id', '0000')
+        return Packet.add_identifiers(pkt, station_id, FOWH32Packet.__name__)
+
+
 class FOWH5Packet(Packet):
     # {"time" : "2019-10-27 14:51:21", "model" : "Fine Offset WH5 sensor", "id" : 48, "temperature_C" : 11.700, "humidity" : 62, "mic" : "CRC"}
 

```

matthewwall, 2024-12-30:
it looks like the WH32 and WH32B are different sensors - 32B has pressure.  so we have a different packet for each.

closed 2024-12-30 by matthewwall