matthewwall/weewx-sdr #165: Multiple radios into single Weewx instance?

vmstan, 2022-08-24:
I'm currently monitoring multiple devices on two frequencies, 433.920M and 915M, hopping between them. It generally works but occasionally I miss packets depending on when devices broadcast them, polling intervals, etc. It seems like the solution to my problem is two SDRs to constantly monitoring each. I think I understand from an rtf_433 command perspective how to trigger a specific device, but how would I configure weewx.conf to poll both?

matthewwall, 2022-08-24:
you probably want to run two weewx instances - one weewx for each sdr, with one sdr looking at 915Mhz and one sdr looking at 433Mhz

vmstan, 2022-08-24:
Ah, wee_reports might be what I've been missing. I saw where you could have multiple instances of Weewx on the same box, but the documentation mentioned it's own database and reports, but nothing about a single report. I dump to an external SQL server on a Synology so I wasn't sure if there was a way to share, but having a unified report from both datasets would be fine. Thanks.

closed 2023-11-03 by vmstan