matthewwall/weewx-sdr #166: ecowitt wh41

farmabep, 2022-09-09:
Good evening everyone
I would like some help to insert in the sensor map inside weewx.conf the right command to acquire the data of
 pm2_5 and pm10_0 and then the battery of the wh41 sensor I tried this way but it does not give me any data:
 # The driver to use
    driver = user.sdr
   log_unknown_sensors = True
   log_unmapped_sensors = True
cmd = rtl_433 -M utc -F json -R 78
[[sensor_map]]
  pm10_0 = pm10_0.142.FOWH0290Packet
        pm2_5 = pm2_5.142.FOWH0290Packet
        wh41_ch1_batt = wh41_ch1_batt.142.FOWH0290Packet
the syslog response was.
Sep 8 21:28:41 raspberrypi weewx [50532] DEBUG user.sdr: punt unrecognized line '{"time": "2022-09-08 19:28:38", "model": "Endoffset-WH0290", " id ": 142," battery_ok ": 0.800," pm2_5_ug_m3 ": 3," estimated_pm10_0_ug_m3 ": 2," family ": 65," unknown1 ": 0," mic ":" CRC "} # 012 '
Sep 8 21:38:42 raspberrypi weewx [50532] DEBUG user.sdr: parse_json: unknown model Fineoffset-WH0290
Sep 8 21:38:42 raspberrypi weewx [50532] DEBUG user.sdr: punt unrecognized line '{"time": "2022-09-08 19:38:38", "model": "Endoffset-WH0290", " id ": 142," battery_ok ": 0.800," pm2_5_ug_m3 ": 2," estimated_pm10_0_ug_m3 ": 2," family ": 65," unknown1 ": 0," mic ":" CRC "} # 012 '
Sep 8 21:38:42 raspberrypi weewx [50532] DEBUG user.sdr: parse_json: unknown model Fineoffset-WH0290
Sep 8 21:38:42 raspberrypi weewx [50532] DEBUG user.sdr: punt unrecognized line '{"time": "2022-09-08 19:38:38", "model": "Endoffset-WH0290", " id ": 142," battery_ok ": 0.800," pm2_5_ug_m3 ": 2," estimated_pm10_0_ug_m3 ": 2," family ": 65," unknown1 ": 0," mic ":" CRC "} # 012 '
Sep 8 21:48:42 raspberrypi weewx [50532] DEBUG user.sdr: parse_json: unknown model Fineoffset-WH0290
Sep 8 21:48:42 raspberrypi weewx [50532] DEBUG user.sdr: punt unrecognized line '{"time": "2022-09-08 19:48:38", "model": "Endoffset-WH0290", " id ": 142," battery_ok ": 0.800," pm2_5_ug_m3 ": 2," estimated_pm10_0_ug_m3 ": 2," family ": 65," unknown1 ": 0," mic ":" CRC "} # 012 '
Sep 8 21:48:42 raspberrypi weewx [50532] DEBUG user.sdr: parse_json: unknown model Fineoffset-WH0290
Sep 8 21:48:42 raspberrypi weewx [50532] DEBUG user.sdr: punt unrecognized line '{"time": "2022-09-08 19:48:38", "model": "Endoffset-WH0290", " id ": 142," battery_ok ": 0.800," pm2_5_ug_m3 ": 2," estimated_pm10_0_ug_m3 ": 2," family ": 65," unknown1 ": 0," mic ":" CRC "} # 012 '
how can i fix it?
I use weewx 4.8.0 (python installation) with weewx sdr updated to the latest release
with raspberry pi4
Thanks in advance

andylittle, 2022-09-09:
try [this](https://github.com/matthewwall/weewx-sdr/blob/08937377d71e1b0899fd5289f5aafd95b17d6a86/bin/user/sdr.py)

farmabep, 2022-09-09:
Sorry Andy .. I made a typo .... Endoffset-WH0290 instead of Endoffsett-WH0290 .... I was wrong ..tomorrow j"ll try

farmabep, 2022-09-09:
Sorry Andy .. I made a typo .... Fineoffset-WH0290 instead of Endoffset-WH0290 .... I was wrong ..

farmabep, 2022-09-09:
Tomorrow j"ll try

andylittle, 2022-09-09:
I assumed it was  IDENTIFIER = "Fineoffset-WH0290" so it should work

farmabep, 2022-09-11:
Thanks,, thanks now it works very well

closed 2022-09-11 by farmabep