matthewwall/weewx-sdr #167: WS68 Support

fankyy, 2022-09-26:
Please add WS68 Support

```
{"time" : "2022-09-26 00:47:08", "model" : "EcoWitt-WS68", "id" : 388, "battery_raw" : 90, "battery_ok" : 1, "lux_raw" : 0, "wind_avg_raw" : 0, "wind_max_raw" : 0, "wind_dir_deg" : 157, "data" : "00 210", "mic" : "CRC"}
{"time" : "2022-09-26 00:47:24", "model" : "EcoWitt-WS68", "id" : 388, "battery_raw" : 90, "battery_ok" : 1, "lux_raw" : 0, "wind_avg_raw" : 0, "wind_max_raw" : 0, "wind_dir_deg" : 157, "data" : "00 210", "mic" : "CRC"}
{"time" : "2022-09-26 00:47:41", "model" : "EcoWitt-WS68", "id" : 388, "battery_raw" : 90, "battery_ok" : 1, "lux_raw" : 0, "wind_avg_raw" : 0, "wind_max_raw" : 0, "wind_dir_deg" : 157, "data" : "00 210", "mic" : "CRC"}
{"time" : "2022-09-26 00:47:57", "model" : "EcoWitt-WS68", "id" : 388, "battery_raw" : 90, "battery_ok" : 1, "lux_raw" : 0, "wind_avg_raw" : 0, "wind_max_raw" : 0, "wind_dir_deg" : 157, "data" : "00 210", "mic" : "CRC"}
```

Thanks a lot!

fankyy

andylittle, 2022-09-26:
What units are battery_raw, wind_avg_raw, wind_max_raw? Do you have the latest version of RTL_433? Is rtl_433 not parsing correctly and outputting the raw values?

Try this in your sdr.py:

`class EcoWittWH68Packet(Packet):
    # This is for a WH68 sensor

    #  {"time" : "2022-09-26 00:47:08", "model" : "EcoWitt-WS68", "id" : 388, "battery_raw" : 90, "battery_ok" : 1, "lux_raw" : 0, "wind_avg_raw" : 0, "wind_max_raw" : 0, "wind_dir_deg" : 157, "data" : "00 210", "mic" : "CRC"}

    IDENTIFIER = "EcoWitt-WH68"

    @staticmethod
    def parse_json(obj):
        pkt = dict()
        pkt['dateTime'] = Packet.parse_time(obj.get('time'))
        pkt['usUnits'] = weewx.METRICWX
        pkt['station_id'] = obj.get('id')
        pkt['battery'] = 0 if obj.get('battery_ok') == 1 else 1
        pkt['luminosity'] = Packet.get_float(obj, 'lux_raw')
        pkt['wind_speed'] = Packet.get_float(obj, 'wind_avg_raw')
        pkt['wind_gust'] = Packet.get_float(obj, 'wind_max_raw')
        pkt['wind_dir'] = Packet.get_float(obj, 'wind_dir_deg')
        return EcoWittWH40Packet.insert_ids(pkt)

    @staticmethod
    def insert_ids(pkt):
        station_id = pkt.pop('station_id', 0)
        return Packet.add_identifiers(pkt, station_id, EcoWittWH68Packet.__name__)`

fankyy, 2022-09-26:
My rtl_433 version is _21.12-113-g631f9f85 branch master at 202207052057

`rtl_433 version 21.12-113-g631f9f85 branch master at 202207052057 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "(null)/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 187 out of 219 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-219 ]`

It seems there is one newer Version from 20220718 (21.12+git20220718 release) but it's only a few days apart so not sure if there are changes for the ws68?

The raw wind numbers are probably the same as in a WH65 or WH69, they seem to be whole numbers always

fankyy, 2022-09-26:
Ok, seems to be a factor of 10:

[image]
[image]
[image]

fankyy, 2022-09-27:
[image]

is there a way to correct this?

matthewwall, 2024-12-30:
added at commit 7a542e8, but the fields look questionable.  i think rtl_433 has stabilized since then...

closed 2024-12-30 by matthewwall

trelaf, 2025-02-27:
The output form rtl_433 has since changed to
```
{
  "time": "2025-02-26 19:33:38",
  "model": "EcoWitt-WS68",
  "id": 1693,
  "battery_raw": 77,
  "battery_ok": 1,
  "light_lux": 1120,
  "wind_avg_m_s": 0.400,
  "wind_max_m_s": 1.500,
  "uvi": 0,
  "wind_dir_deg": 157,
  "data": "205",
  "mic": "CRC"
}
```
Im using this updated mapping

```
class EcoWittWS68Packet(Packet):
    # This is for a WS68 wind/solar sensor

    IDENTIFIER = "EcoWitt-WS68"

    @staticmethod
    def parse_json(obj):
        sensor_id = obj.get('id')
        pkt = dict()
        pkt['dateTime'] = Packet.parse_time(obj.get('time'))
        pkt['usUnits'] = weewx.METRICWX
        pkt['wind_speed'] = Packet.get_float(obj, 'wind_avg_m_s')
        pkt['wind_gust'] = Packet.get_float(obj, 'wind_max_m_s')
        pkt['wind_dir'] = Packet.get_float(obj, 'wind_dir_deg')
        pkt['light'] = Packet.get_float(obj, 'light_lux')
        pkt['uv_index'] = Packet.get_float(obj, 'uvi')
        pkt['battery'] = Packet.get_battery(obj)
        pkt = Packet.add_identifiers(pkt, sensor_id, EcoWittWS68Packet.__name__)
        return pkt
```
`battery_raw` may be a percentage, but I'm not sure. Not mapped.