matthewwall/weewx-sdr #169: WH3085

mfraser, 2022-11-01:
Received a WH3085 today which identifies itself as a Fineoffset-WHx080.
The usual WHx080 data is parsed correctly, but the data that would normally be parsed if it was a WH3080 is not.

Can I just copy the lines from WH3080 into WHx080?

"time" : "2022-11-01 18:23:01",
"model" : "Fineoffset-WHx080",
"subtype" : 2,
"uv_sensor_id" : 113,
"uv_status" : "OK",
"uv_index" : 0,
"lux" : 0.000,
"wm" : 0.000,
"mic" : "CRC"
'{"time" : "2022-11-01 18:23:01",
"model" : "Fineoffset-WHx080",
"subtype" : 2,
"uv_sensor_id" : 113,
"uv_status" : "OK",
"uv_index" : 0,
"lux" : 0.000,
"wm" : 0.000,
"mic" : "CRC"
"time" : "2022-11-01 18:23:04",
"model" : "Fineoffset-WHx080",
"time" : "2022-11-01 18:23:17",
"model" : "Fineoffset-WHx080", 
"subtype" : 0,
"id" : 113,
"battery_ok" : 1,
"temperature_C" : 21.900,
"humidity" : 67,
"wind_dir_deg" : 135,
"wind_avg_km_h" : 0.000,
"wind_max_km_h" : 0.000,
"rain_mm" : 0.300,
"mic" : "CRC"
parsed: {
'dateTime': 1667326997,
'usUnits': 16,
'msg_type.113.FOWHx080Packet': 0,
'temperature.113.FOWHx080Packet': 21.9,
'humidity.113.FOWHx080Packet': 67.0,
'wind_dir.113.FOWHx080Packet': 135.0,
'wind_speed.113.FOWHx080Packet': 0.0,
'wind_gust.113.FOWHx080Packet': 0.0,
'rain_total.113.FOWHx080Packet': 0.03,
'battery.113.FOWHx080Packet': 0,
'signal_type.113.FOWHx080Packet': 0,
'hours.113.FOWHx080Packet': None,
'minutes.113.FOWHx080Packet': None,
'seconds.113.FOWHx080Packet': None,
'year.113.FOWHx080Packet': None,
'month.113.FOWHx080Packet': None,
'day.113.FOWHx080Packet': None}

mfraser, 2022-11-02:
I've tried copying the lines:
```
pkt['uv_index'] = Packet.get_float(obj, 'uv_index')
        pkt['luminosity'] = Packet.get_float(obj, 'lux')
        pkt['radiation'] = Packet.get_float(obj, 'wm')
        pkt['illumination'] = Packet.get_float(obj, 'fc')
        pkt['uv_status'] = 0 if obj.get('uv_status') == 'OK' else 1
```
from FOWH3080, but that gives me lines like:
```
[out: ['{"time" : "2022-11-02 15:06:00", "model" : "Fineoffset-WHx080", "subtype" : 2, "uv_sensor_id" : 113, "uv_status" : "OK", "uv_index" : 0, "lux" : 821.000, "wm" : 1.202, "mic" : "CRC"}\n']
parsed: {'dateTime': 1667401560, 'usUnits': 16, 'msg_type.0000.FOWHx080Packet': 2, 'temperature.0000.FOWHx080Packet': None, 'humidity.0000.FOWHx080Packet': None, 'wind_dir.0000.FOWHx080Packet': None, 'wind_speed.0000.FOWHx080Packet': None, 'wind_gust.0000.FOWHx080Packet': None, 'uv_index.0000.FOWHx080Packet': 0.0, 'luminosity.0000.FOWHx080Packet': 821.0, 'radiation.0000.FOWHx080Packet': 1.202, 'uv_status.0000.FOWHx080Packet': 0, 'battery.0000.FOWHx080Packet': 1, 'signal_type.0000.FOWHx080Packet': 0, 'hours.0000.FOWHx080Packet': None, 'minutes.0000.FOWHx080Packet': None, 'seconds.0000.FOWHx080Packet': None, 'year.0000.FOWHx080Packet': None, 'month.0000.FOWHx080Packet': None, 'day.0000.FOWHx080Packet': None}]

```
Where the ID is 0000.

andylittle, 2026-02-06:
can you verify the latest sdr.py driver is working? the code looks to match your example

Andy

hilman2, 2026-10-21:
Verified current WHx080 subtype-2 handling with your UV sensor ID 113 and zero UV reading. https://github.com/weewx-packages/weewx-sdr/releases/tag/v0.99.0 includes the regression fixture; the identifiers and radiation value are retained.