matthewwall/weewx-sdr #177: WS90 Support

fankyy, 2023-05-18:
The WS90 protocol was recently added in rtl_433:

[https://github.com/merbanan/rtl_433/pull/2448](url)

Please add to the weewx-sdr driver :)

**output:**
_rtl_433 -f 868.3M -M time:utc -s 300k -R 244_
```_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-05-18 22:57:35
model     : Fineoffset-WS90                        ID        : 00273f
Battery   : 1.000        Battery Voltage: 3280 mV  Temperature: 9.7 C        Humidity  : 85 %          Wind direction: 108       Wind speed: 1.6 m/s
Gust speed: 1.7 m/s      UVI       : 0.0           Light     : 0.0 lux       Flags     : 89            Total Rain: 874.7 mm      Supercap Voltage: 5.0 V
Extra Data: fe47222b3277b0ff7ff8000077             Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-05-18 22:57:53
model     : Fineoffset-WS90                        ID        : 00273f
Battery   : 1.000        Battery Voltage: 3280 mV  Temperature: 9.7 C        Humidity  : 85 %          Wind direction: 114       Wind speed: 1.6 m/s
Gust speed: 1.7 m/s      UVI       : 0.0           Light     : 0.0 lux       Flags     : 89            Total Rain: 874.7 mm      Supercap Voltage: 5.0 V
Extra Data: fe47222b3277b0ff9ff9000077             Integrity : CRC
```

**output with level:**
_rtl_433 -f 868.3M -M time:utc -M protocol -M level -s 300k -R 244_
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-05-18 22:44:10                    Protocol  : 244
model     : Fineoffset-WS90                        ID        : 00273f
Battery   : 1.000        Battery Voltage: 3280 mV  Temperature: 10.1 C       Humidity  : 82 %          Wind direction: 105       Wind speed: 1.4 m/s
Gust speed: 1.4 m/s      UVI       : 0.0           Light     : 0.0 lux       Flags     : 89            Total Rain: 874.7 mm      Supercap Voltage: 5.0 V
Extra Data: fe47222b3277b0ff9ff8000077             Integrity : CRC
Modulation: FSK          Freq1     : 868.3 MHz     Freq2     : 868.4 MHz
RSSI      : -0.1 dB      SNR       : 22.6 dB       Noise     : -22.7 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-05-18 22:44:36                    Protocol  : 244
model     : Fineoffset-WS90                        ID        : 00273f
Battery   : 1.000        Battery Voltage: 3280 mV  Temperature: 10.1 C       Humidity  : 82 %          Wind direction: 103       Wind speed: 1.5 m/s
Gust speed: 1.5 m/s      UVI       : 0.0           Light     : 0.0 lux       Flags     : 89            Total Rain: 874.7 mm      Supercap Voltage: 5.0 V
Extra Data: fe47222b3277b0ffaff8000077             Integrity : CRC
Modulation: FSK          Freq1     : 868.3 MHz     Freq2     : 868.4 MHz
RSSI      : -0.1 dB      SNR       : 24.9 dB       Noise     : -25.1 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
```

**json output:**
_rtl_433 -f 868.3M -M time:utc -s 300k -F json -R 244_
`
{"time" : "2023-05-18 22:55:06", "model" : "Fineoffset-WS90", "id" : 10047, "battery_ok" : 1.000, "battery_mV" : 3280, "temperature_C" : 9.700, "humidity" : 85, "wind_dir_deg" : 86, "wind_avg_m_s" : 0.900, "wind_max_m_s" : 1.400, "uvi" : 0.000, "light_lux" : 0.000, "flags" : 137, "rain_mm" : 874.700, "supercap_V" : 5.000, "data" : "fe47222b3277b0ff7ff8000077", "mic" : "CRC"}
{"time" : "2023-05-18 22:55:15", "model" : "Fineoffset-WS90", "id" : 10047, "battery_ok" : 1.000, "battery_mV" : 3280, "temperature_C" : 9.700, "humidity" : 86, "wind_dir_deg" : 87, "wind_avg_m_s" : 1.000, "wind_max_m_s" : 1.200, "uvi" : 0.000, "light_lux" : 0.000, "flags" : 137, "rain_mm" : 874.700, "supercap_V" : 5.000, "data" : "fe47222b3277b0ff8ff7000077", "mic" : "CRC"}
`

**json output with level:**
_rtl_433 -f 868.3M -M time:utc -M protocol -M level -s 300k -F json_ -R244
`{"time" : "2023-05-18 22:46:30", "protocol" : 244, "model" : "Fineoffset-WS90", "id" : 10047, "battery_ok" : 1.000, "battery_mV" : 3280, "temperature_C" : 10.100, "humidity" : 83, "wind_dir_deg" : 130, "wind_avg_m_s" : 2.000, "wind_max_m_s" : 2.000, "uvi" : 0.000, "light_lux" : 0.000, "flags" : 137, "rain_mm" : 874.700, "supercap_V" : 5.000, "data" : "fe47222b3277b0ff8ff7000077", "mic" : "CRC", "mod" : "FSK", "freq1" : 868.305, "freq2" : 868.385, "rssi" : -0.120, "snr" : 22.247, "noise" : -22.367}`

Thanks a lot!
fankyy

mfraser, 2023-05-21:
Have a look at the fork here https://github.com/jpochmara/weewx-sdr which has added support for the WS90.

fankyy, 2023-08-26:
Oh I was late to the game, didn't know about that fork and that it was already added - works perfect!

Does anyone know what's the easiest way to add the snr/rssi/noise to sensor map and if it can be linked to pre-existing labels?

matthewwall, 2024-12-30:
these have been incorporated into weewx-sdr by v0.95 with freq, rssi, etc. added to other sensors by commit 58ba55f

closed 2024-12-30 by matthewwall