matthewwall/weewx-sdr #179: unknown model Bresser-7in1

farmabep, 2023-06-11:
Good evening everyone. I would like someone to please help me to insert in the sensor map and in the sdr.py file the instructions for a new Bresser 7in 1 station. The sensor from which I would like to get the data is called 7803210 4cast bresser and it works at 868M
I use the weewx-sdr applied to a raspberry pi 4 and I capture the signals from the sensor thanks to a Noolec Nesdr pen
Here are the results from the weewx log file:
Jun 11 18:52:28 raspberrypi weewx[2631] DEBUG user.sdr: punt unrecognized line '{"time" : "2023-06-11 16:52:25", "model" : "Bresser-7in1", " id" : 50437, "temperature_C" : 24.000, "humidity" : 65, "wind_max_m_s" : 0.800, "wind_avg_m_s" : 0.800, "wind_dir_deg" : 48, "rain_mm" : 3.500, "light_klx" : 10.357, "light_lux" : 10357.000, "uv" : 0.900, "battery_ok" : 1, "mic" : "CRC"}#012'

this is the result of the command sudo PYTHONPATH=bin python bin/user/sdr.py --cmd="rtl_433 -f 868M -M utc -F json"
out: ['{"time" : "2023-06-11 17:09:05", "model" : "Bresser-7in1", "id" : 50437, "temperature_C" : 23.500, "humidity" : 67, "wind_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 102, "rain_mm" : 3.500, "light_klx" : 8.592, "light_lux" : 8592.000, "uv" : 1.000, "battery_ok" : 1, "mic " : "CRC"}\n']
I insert if it was useful the first part of the weewx.conf file:
      [SDR]
    # This section is for the software-defined radio driver.
    
    # The driver to use
    driver = user.sdr
     log_unknown_sensors = True
     log_unmapped_sensors = True
    cmd = rtl_433 -f 868M -M utc -F json -R 173
    [[sensor_map]]
        outTemp = temperature.50437.Bresser7in1Packet (?????????)
Grazie per  la vs disponibilita'

andylittle, 2023-06-12:
Try this updated driver
https://github.com/andylittle/weewx-sdr-1/blob/master/bin/user/sdr.py    

yes Bresser7in1Packet. Please verify the values are correct.

temperature.50437.Bresser7in1Packet
humidity.50437.Bresser7in1Packet
wind_gust.50437.Bresser7in1Packet
wind_speed.50437.Bresser7in1Packet
wind_dir.50437.Bresser7in1Packet
rain_total.50437.Bresser7in1Packet
lux.50437.Bresser7in1Packet
uv.50437.Bresser7in1Packet
battery.50437.Bresser7in1Packet

Andy

farmabep, 2023-06-18:
hello Andy, first of all thanks for the new sdy.py
I tried for a week the new configuration with the new file you sent me. Everything works except for the rainfall and rainrate values that appear on ///home/weewx/public_html/index.html which are busted and appear even without rain (ghost values), As you can see from this summary the rain values remain constant and do not increase
Jun 18 13:03:02 raspberrypi weewx[1358] DEBUG user.sdr: packet={'outTemp': 29.9, 'outHumidity': 38.0, 'windDir': 0.0, 'windSpeed': 1.1, 'windGust': 1.1, 'rain': 14.9, 'dateTime': 1687086179, 'usUnits': 17}
Jun 18 13:03:15 raspberrypi weewx[1358] DEBUG user.sdr: packet={'outTemp': 29.9, 'outHumidity': 39.0, 'windDir': 0.0, 'windSpeed': 1.0, 'windGust': 1.0, 'rain': 14.9, 'dateTime': 1687086192, 'usUnits': 17}
Jun 18 13:03:27 raspberrypi weewx[1358] DEBUG user.sdr: packet={'outTemp': 29.9, 'outHumidity': 41.0, 'windDir': 0.0, 'windSpeed': 0.0, 'windGust': 0.0, 'rain': 14.9, 'dateTime': 1687086204, 'usUnits': 17}
Jun 18 13:03:40 raspberrypi weewx[1358] DEBUG user.sdr: packet={'outTemp': 29.9, 'outHumidity': 40.0, 'windDir': 0.0, 'windSpeed': 0.8, 'windGust': 0.8, 'rain': 14.9, 'dateTime': 1687086217, 'usUnits': 17}

In the log file (sudo tail -f /var/log/syslog) the rainrate values increase dramatically without the presence of rain. Here are short lines of text:
LOOP: 2023-06-18 13:11:44 CEST (1687086704) 'altimeter': 'None', 'appTemp': '88.70964042909002', 'barometer': 'None', 'cloudbase': '6537.065355698671', 'dateTime ': '1687086704', 'dewpoint': '61.675453203191104', 'ET': 'None', 'heatindex': '86.55847921471613', 'humidex': '95.06074786321693', 'inDewpoint': 'None', 'maxSolarRad': '994.6179794017675', 'outHumidity': '44.0', 'outTemp': '86.18', 'pressure': 'None', 'rain': '0.5866141732283465', 'rainRate': '105.59055118110247', 'usUnits': '1 ', 'windchill': '86.18', 'windDir': '258.0', 'windGust': '2.460636037133235', 'windrun': 'None', 'windSpeed': '2.460636037133235'
LOOP: 2023-06-18 13:11:57 CEST (1687086717) 'altimeter': 'None', 'appTemp': '88.20438404594957', 'barometer': 'None', 'cloudbase': '6835.736633820963', 'dateTime ': '1687086717', 'dewpoint': '60.36129957945302', 'ET': 'None', 'heatindex': '86.08832131063602', 'humidex': '94.18907222422597', 'inDewpoint': 'None', 'maxSolarRad': '994.6202791175209', 'outHumidity': '42.0', 'outTemp': '86.18', 'pressure': 'None', 'rain': '0.5866141732283465', 'rainRate': '107.93700787401586', 'usUnits': '1 ', 'windchill': '86.18', 'windDir': '251.0', 'windGust': '2.460636037133235', 'windrun': 'None', 'windSpeed': '2.460636037133235'
LOOP: 2023-06-18 13:12:22 CEST (1687086742) 'altimeter': 'None', 'appTemp': '88.56919149996878', 'barometer': 'None', 'cloudbase': '6994.150453027647', 'dateTime ': '1687086742', 'dewpoint': '59.844278774943604', 'ET': 'None', 'heatindex': '86.0814679702721', 'humidex': '94.03611093698922', 'inDewpoint': 'None', 'maxSolarRad': '994.6225276946826', 'outHumidity': '41.0', 'outTemp': '86.36', 'pressure': 'None', 'rain': '0.5866141732283465', 'rainRate': '110.28346456692925', 'usUnits': '1 ', 'windchill': '86.36', 'windDir': '252.0', 'windGust': '1.7895534815514436', 'windrun': 'None', 'windSpeed': '1.7895534815514436'
LOOP: 2023-06-18 13:12:34 CEST (1687086754) 'altimeter': 'None', 'appTemp': '88.82534718289408', 'barometer': 'None', 'cloudbase': '6689.038062666471', 'dateTime ': '1687086754', 'dewpoint': '61.18677329253278', 'ET': 'None', 'heatindex': '86.54684603740799', 'humidex': '94.91230248309932', 'inDewpoint': 'None', 'maxSolarRad': '994.6226299019414', 'outHumidity': '43.0', 'outTemp': '86.36', 'pressure': 'None', 'rain': '0.5866141732283465', 'rainRate': '112.62992125984265', 'usUnits': '1 ', 'windchill': '86.36', 'windDir': '114.0', 'windGust': '2.2369418519393043', 'windrun': 'None', 'windSpeed': '2.2369418519393043'
LOOP: 2023-06-18 13:12:47 CEST (1687086767) 'altimeter': 'None', 'appTemp': '88.06311365850613', 'barometer': 'None', 'cloudbase': '7151.630557326211', 'dateTime ': '1687086767', 'dewpoint': '59.15136631602992', 'ET': 'None', 'heatindex': '85.86075147619206', 'humidex': '93.5985038264838', 'inDewpoint': ' None', 'maxSolarRad': '994.6216810436254', 'outHumidity': '40.0', 'outTemp': '86.36', 'pressure': 'None', 'rain': '0.5866141732283465', 'rainRate': '114.97637795275604', 'usUnits': '1 ', 'windchill': '86.36', 'windDir': '246.0', 'windGust': '2.2369418519393043', 'windrun': 'None', 'windSpeed': '2.2369418519393043'
LOOP: 2023-06-18 13:12:59 CEST (1687086779) 'altimeter': 'None', 'appTemp': '88.19119149996877', 'barometer': 'None', 'cloudbase': '6994.150453027647', 'dateTime ': '1687086779', 'dewpoint': '59.84427877494360
I tried inserting ferrite rings into the usb cable hoping there was electromagnetic interference but it didn't solve the problem
thanks for your time'....

farmabep, 2023-06-18:
I add the functioning of data capture by the nesdr stick by inserting a small log here:
pi@raspberrypi:~$rtl_433 -f 868M

time : 2023-06-18 13:20:17
model : Bresser-7in1 id : 50437
Temperature: 30.1 C Humidity : 41 Wind Gust : 0.7 m/s Wind Speed: 0.7 m/s
Direction : 193 Rain : 14.9 mm Light : 109.518 klx Light : 109518.000 lux
UV Index : 9.4 Battery : 1 Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time : 2023-06-18 13:20:29
model : Bresser-7in1 id : 50437
Temperature: 30.1 C Humidity : 40 Wind Gust : 0.9 m/s Wind Speed: 0.9 m/s
Direction : 196 Rain : 14.9 mm Light : 109.699 klx Light : 109699.000 lux
UV Index : 9.2 Battery : 1 Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time : 2023-06-18 13:20:42
model : Bresser-7in1 id : 50437
Temperature: 30.1 C Humidity : 39 Wind Gust : 0.7 m/s Wind Speed: 0.7 m/s
Direction : 196 Rain : 14.9 mm Light : 109.745 klx Light : 109745.000 lux
UV Index : 9.8 Battery : 1 Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time : 2023-06-18 13:21:19
model : Bresser-7in1 id : 50437
Temperature: 30.1 C Humidity : 42 Wind Gust : 1.8 m/s Wind Speed: 1.6 m/s
Direction : 205 Rain : 14.9 mm Light : 110.054 klx Light : 110054.000 lux
UV Index : 9.7 Battery : 1 Integrity : CRC
the rainfall values, on the other hand, remain constant also in the subsequent logs without variations

andylittle, 2023-06-19:
try this in your config

rain_total = rain_total.50437.Bresser7in1Packet
    [[deltas]]
        rain = rain_total

farmabep, 2023-06-23:
Thanks Andy, the correction you suggested on the rain value was perfect. One last thing I ask you please: what is the command that allows you to see all the packets of all the sensors to then insert in the sensor map??
I used the very comfortable command: sudo PYTHONPATH=/home/weewx/bin python /home/weewx/bin/user/sdr.py, I see all the 433mhz sensor packets but I can't see the 868mhz sensor packets mhz
here is a short log:
out: ['{"time" : "2023-06-23 17:31:32", "model" : "Fineoffset-WH65B", "id" : 14, "battery_ok" : 1, "temperature_C" : 21.500, "humidity" : 94, "wind_dir_deg" : 39, "wind_avg_m_s" : 0.000, "wind_max_m_s" : 0.000, "rain_mm" : 28.194, "uv" : 74, "uvi" : 0, "light_lux" : 9393.000, "mic " : "CRC"}\n', '{"time" : "2023-06-23 17:31:34", "model" : "Hideki-Wind", "id" : 8, "channel" : 4 , "battery_ok" : 1, "temperature_C" : 27.900, "wind_avg_mi_h" : 0.000, "wind_max_mi_h" : 0.000, "wind_approach" : 0, "wind_dir_deg" : 202.500, "mic" : "CRC"}\n', ' {"time" : "2023-06-23 17:31:34", "model" : "Hideki-Wind", "id" : 8, "channel" : 4, "battery_ok" : 1, "temperature_C" : 27.900, "wind_avg_mi_h" : 0.000, "wind_max_mi_h" : 0.000, "wind_approach" : 0, "wind_dir_deg" : 202.500, "mic" : "CRC"}\n', '{"time" : "2023-06-23 17:31:34", "model" : "Hideki-Wind", "id" : 8, "channel" : 4, "battery_ok" : 1, "temperature_C" : 27.900, "wind_avg_mi_h" : 0.000, "wind_max_mi_h" : 0.000, "wind_approach" : 0, "wind_dir_deg" : 202.500, "mic" : "CRC"}\n']
parsed: {'dateTime': 1687541492, 'usUnits': 17, 'temperature.14.FOWH65BAltPacket': 21.5, 'humidity.14.FOWH65BAltPacket': 94.0, 'wind_dir.14.FOWH65BAltPacket': 39.0, 'wind_speed.14. FOWH65BAltPacket': 0.0, 'wind_gust.14.FOWH65BAltPacket': 0.0, 'rain_total.14.FOWH65BAltPacket': 28.194, 'uv.14.FOWH65BAltPacket': 74.0, 'uv_index.14.FOWH65BAltPacket': 0.0, 'light. 14. FOWH65BAltPacket': 9393.0, 'battery.14.FOWH65BAltPacket': 0}
parsed: {'dateTime': 1687541494, 'usUnits': 16, 'temperature.4:None.HidekiWindPacket': 27.9, 'wind_speed.4:None.HidekiWindPacket': 0.0, 'wind_dir.4:None.HidekiWindPacket': 202.5 , 'wind_gust.4:None.HidekiWindPacket': 0.0, 'battery.4:None.HidekiWindPacket': 0}
parsed: {'dateTime': 1687541494, 'usUnits': 16, 'temperature.4:None.HidekiWindPacket': 27.9, 'wind_speed.4:None.HidekiWindPacket': 0.0, 'wind_dir.4:None.HidekiWindPacket': 202.5 , 'wind_gust.4:None.HidekiWindPacket': 0.0, 'battery.4:None.HidekiWindPacket': 0}
parsed: {'dateTime': 1687541494, 'usUnits': 16, 'temperature.4:None.HidekiWindPacket': 27.9, 'wind_speed.4:None.HidekiWindPacket': 0.0, 'wind_dir.4:None.HidekiWindPacket': 202.5 , 'wind_gust.4:None.HidekiWindPacket': 0.0, 'battery.4:None.HidekiWindPacket': 0}

I'm also asking you because I'd like to learn to see and interpret the packets of other sensors that I own so as not to disturb you further and fill the sensor map by myself.
Thanks for your valuable suggestions

andylittle, 2023-06-26:
just specify the rtl cmd, I am not sure if this frequency is correct. run rtl_433  -h to see the options.
 
`PYTHONPATH=/home/weewx/bin python3 sdr.py --cmd="rtl_433 -M utc -F json -f 868000000" `

magnushult20, 2023-12-15:
here is a output from the starting the weewx   . It claims unmapped data all the way.
Dec 15 13:47:17 magnus-OptiPlex-9020M weewx[33200] INFO weewx.engine: Starting main packet loop.

Dec 15 13:47:30 magnus-OptiPlex-9020M weewx[33200] INFO user.sdr: unmapped: {'dateTime': 1702644446, 'usUnits': 17, 'temperature.17041.Bresser7in1Packet': -1.6, 'humidity.17041.Bresser7in1Packet': 96.0, 'wind_gust.17041.Bresser7in1Packet': 0.8, 'wind_speed.17041.Bresser7in1Packet': 0.8, 'wind_dir.17041.Bresser7in1Packet': 240.0, 'rain_total.17041.Bresser7in1Packet': 105.6, 'lux.17041.Bresser7in1Packet': None, 'uv.17041.Bresser7in1Packet': 0.0, 'battery.17041.Bresser7in1Packet': 0}
  and extract from the weewx.conf file

Dec 15 13:47:55 magnus-OptiPlex-9020M weewx[33200] INFO user.sdr: unmapped:     ##  Direct testing command:
    # PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/sdr.py   cmd = rtl_433 -M utc -F json -f 868M -R 173
    cmd = rtl_433 -M utc -F json -f 868M
####-R 173

        [[sensor map]]
        outTemp = temperature.17041.Bresser7in1Packet
        outHumidity = humidity.17041.Bresser7in1Packet
        windGust = wind_gust.17041.Bresser7in1Packet
        windSpeed = wind_avg_m_s.17041.Bresser7in1Packet
        wind_Dir = wind_dir.17041.Bresser7in1Packet
        rain_total = rain_total.17041.Bresser7in1Packet
        lux = lux.17041.Bresser7in1Packet
        UV = uv.17041.Bresser7in1Packet
        outTempBatteryStatus = battery.17041.Bresser7in1Packet
        [[deltas]]
        rain = rain_total

magnus

closed 2024-02-14 by farmabep