matthewwall/weewx-sdr #181: not getting data

promy, 2023-11-10:
I'm unable to get data to weewx. 
A bit more info
I installed the driver cfr the readme
When i run  rlt_433 -M utc -F json   i am seeing the data
i edited the weewx with the sensor map (for testing i commented out the wind/rain)
[[sensor_map]]
        outBatteryStatus = battery.1:9.HidekiTS04Packet
        outHumidity = humidity.1:9.HidekiTS04Packet
        outTemp = temperature.1:9.HidekiTS04Packet

but the log shows nothing from weewx after 
INFO weewx.engine: Starting main packet loop.

switching to the TE923 and i'm getting the data (but the problem is the tfa sinus is loosing the sensors after 10minutes due to inferference from the usb (known problem it seems)

promy, 2023-11-10:
after a few hours i'm getting the error "user.sdr: parse_json: unknown model Microchip-HCS200" and "DEBUG user.sdr: punt unrecognized line '{"time" : "2023-11-10 13:15:29", "model" : "Microchip-HCS200", "id" : "04CF7AA", "battery_ok" : 0, "button" : 1, "learn" : 0, "repeat" : 1, "encrypted" : "943B3E5A"}#012'"

andylittle, 2023-11-10:
post some of the output of  rlt_433 -M utc -F json

promy, 2023-11-10:
here are some of the values

{"time" : "2023-11-10 14:41:47", "model" : "Hideki-TS04", "id" : 6, "channel" : 1, "battery_ok" : 1, "temperature_C" : 7.600, "humidity" : 84, "mic" : "CRC"}
{"time" : "2023-11-10 14:41:47", "model" : "Hideki-TS04", "id" : 6, "channel" : 1, "battery_ok" : 1, "temperature_C" : 7.600, "humidity" : 84, "mic" : "CRC"}
{"time" : "2023-11-10 14:41:47", "model" : "Hideki-TS04", "id" : 6, "channel" : 1, "battery_ok" : 1, "temperature_C" : 7.600, "humidity" : 84, "mic" : "CRC"}
{"time" : "2023-11-10 14:41:53", "model" : "Hideki-TS04", "id" : 0, "channel" : 2, "battery_ok" : 1, "temperature_C" : 22.000, "humidity" : 60, "mic" : "CRC"}
{"time" : "2023-11-10 14:41:53", "model" : "Hideki-TS04", "id" : 0, "channel" : 2, "battery_ok" : 1, "temperature_C" : 22.000, "humidity" : 60, "mic" : "CRC"}
{"time" : "2023-11-10 14:41:53", "model" : "Hideki-TS04", "id" : 0, "channel" : 2, "battery_ok" : 1, "temperature_C" : 22.000, "humidity" : 60, "mic" : "CRC"}
{"time" : "2023-11-10 14:42:05", "model" : "Hideki-Rain", "id" : 0, "channel" : 4, "battery_ok" : 1, "rain_mm" : 2255.400, "mic" : "CRC"}
{"time" : "2023-11-10 14:42:05", "model" : "Hideki-Rain", "id" : 0, "channel" : 4, "battery_ok" : 1, "rain_mm" : 2255.400, "mic" : "CRC"}
{"time" : "2023-11-10 14:42:06", "model" : "Hideki-Rain", "id" : 0, "channel" : 4, "battery_ok" : 1, "rain_mm" : 2255.400, "mic" : "CRC"}

promy, 2023-11-10:
an extra comment: in the log it shows up if my garage door remote is pressed: "weewx[16535] DEBUG user.sdr: parse_json: unknown model Microchip-HCS200"
so something is happening i guess

promy, 2023-11-11:
I found the problem: the example suggests  outBatteryStatus = battery.1:9.HidekiTS04Packet  
But the 9 is the id.... mine is another number, after changing this, it was solved ;-) 
maybe something to add in the readme?

matthewwall, 2024-12-30:
andylittle's branch has been merged into weewx-sdr, including this fix

closed 2024-12-30 by matthewwall