matthewwall/weewx-sdr #183: Stuck on Step #2

n3eds, 2024-02-18:
All seems to be installed so far.  Already had all the RTL-SDR Tools installed and RTL-433 installed and working fine.  Installed WeeWX from teh docs on the website fine and seems to be running on the simulator.  Running a PPI4B 8gb with all latest updates.  Seems the two wee_extension and wee_config commands do not exist to run them.  Command not found...

2) install the driver

sudo wee_extension --install weewx-sdr.zip

3) configure the driver

sudo wee_config --reconfigure --driver=user.sdr --no-prompt

BrightmanRandy, 2024-02-19:
They changed the command on version 5.0 to:

sudo weectl extension install weewx-sdr.zip

You will also have to add weewx user to plugdev group as outlined here. https://groups.google.com/g/weewx-user/c/7CIRRoo6Qdk/m/8dwn6EBiAQAJ

n3eds, 2024-02-20:
You know, this is a convoluted disaster to say the least.  I have countess
apps running just fine on many RPI's utilizing SDR's and never have I ran
into such a mess with conflicting instructions.  All I want is a simple WX
station running on an SDR, that's all.   Maybe I'll revisit this in a few
years when all the apps like rtl_433 (which BTW works fine on all of my
machines and SDR's) and the instructions are all in one, updated place that
work.  Sorry, not with the hassle.

unmanagedtn, 2024-03-23:
https://groups.google.com/g/weewx-user/c/svsbOxWvPZc

See the first link and the howto.txt in the github repo

n3eds, 2024-03-23:
Thanks!  I will definitely give this a try.

unmanagedtn, 2024-03-24:
Honestly I have decided to shift to a different solution from here
https://www.scaledinstruments.com/shop/shop-by-category/data-loggers/wifilogger2/

unmanagedtn, 2024-03-24:
and also if you want to stick with PI stuff then here is the way to go for
that ... until these projects are more cohesive then I am also going to see
alternatives...

https://www.scaledinstruments.com/shop/shop-by-category/data-loggers/meteo-pi/

matthewwall, 2024-12-29:
closing this, as it was apparently resolved.  

for reference, here are weewx-sdr instructions, all in one place:

https://github.com/weewx/weewx/wiki/sdr-rpi-recipe

for those using weewx v4, here are the semanitcs:

https://weewx.com/docs/4.10/utilities.htm#wee_extension_utility

for those using weewx v5, here are the semantics:

https://weewx.com/docs/5.1/utilities/weectl-extension/

documentation for every version of weewx:

https://weewx.com/docs.html

closed 2024-12-29 by matthewwall