matthewwall/weewx-sdr #185: Trouble capturing Data from Acurite Atlas 7in1 and WH32B reliably

agc346, 2024-04-21:
I have tried frequency hopping from 433 to 915 with no luck. Whatever frequency is specified first in the command works, but the second does not. I have tried two SDRs with mixed success, however I am unable to get two SDRs to feed information into weewx. A couple commands I have tried are:

rtl_433 -M utc -F json -f 433.92M -f 914.98M -H 35 -R 40 -R 78
rtl_433 -d 0 -M utc -F json -R 40 & rtl_433 -d 1 -M utc -F json -f 914980000 -R 78

These work, although I am unable to combine them to receive data.
rtl_433 -d 0 -M utc -F json -R 40
rtl_433 -d 1 -M utc -F json -f 914980000 -R 78

I would like to have frequency hopping working so I only need one sdr.

sandeen, 2024-05-27:
If the rtl_433 commands are the problem, I think you need to raise the issue with the project at https://github.com/merbanan/rtl_433 rather than here?

agc346, 2024-07-17:
Thank You. I have added the issue there.

matthewwall, 2024-12-30:
i never had any luck getting a single sdr to sample multiple frequencies - always had to run a separate weewx instance for each sdr on each frequency.  please let us know if you figure out how to do it!

matthewwall, 2024-12-30:
for the record, this was the response in the rtl_433 issue

> Do you see the notice about hopping every 35 seconds?
> With current rtl_433 you will need to add the kv or log output to get messages when using another 
> output like json.
>
> Maybe try just `rtl_433 -v -f 433.92M -f 914.98M -H 35 -R 40 -R 78 -M utc` (note that order can be 
> important and input should go first, outputs last)

https://github.com/merbanan/rtl_433/issues/3000

closed 2024-12-30 by matthewwall