matthewwall/weewx-sdr #186: Fix docs for rain total in [[sensor_map]] ?

sandeen, 2024-05-26:
Hi Matthew - 

I think that upstream weewx might need `rainTotal` rather than `rain_total` for the sensor map examples:
```
    [[sensor_map]]
        windDir = wind_dir.0BFA.Acurite5n1Packet
        windSpeed = wind_speed.0BFA.Acurite5n1Packet
        outTemp = temperature.0BFA.Acurite5n1Packet
        outHumidity = humidity.0BFA.Acurite5n1Packet
        rain_total = rain_total.0BFA.Acurite5n1Packet
        inTemp = temperature.24A4.AcuriteTowerPacket
        inHumidity = humidity.24A4.AcuriteTowerPacket
```

Not sure if this is a recent change, but i wasn't getting rain reported Weather Underground until I made that change.
I couldn't tell how to find what all the expected weewx names are, though ...?

sandeen, 2024-05-27:
Ok now I'm not sure about this, but still not sure how to find the definitive list of database names ...

matthewwall, 2024-12-30:
weewx wants an observation 'rain' that is the amount of rain in the last period.

weewx-sdr does two things: (1) map the sensor to a variable, e.g., 'rain_total', and (2) indicate which variables are cumulative, e.g., 'rain_total' is a cumulative measure.  (1) is done in the sensor_map, (2) is done in the deltas section.

i added an explanation to the readme at commit 8abc489

closed 2024-12-30 by matthewwall