matthewwall/weewx-sdr #192: Help for sensor map EMAX - EM3551H

Alpha-SG31, 2024-10-25:
Hi, 
I would like help to create the sensor map for my station and also for other users if is possible. 
This chip is embedded in several stations but with different brands. 
here is my rtl433 output it is an EMAX - EM3551H sensor.
Thx U

{"time" : "2024-10-25 17:51:33", "model" : "Emax-EM3551H", "id" : 1001, "channel" : 4, "battery_ok" : 1, "temperature_F" : 55.700, "humidity" : 95, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "wind_dir_deg" : 169, "rain_mm" : 0.000, "mic" : "CHECKSUM"}

Alpha-SG31, 2024-11-03:
I finally managed to decode my station. 
I searched the docs and certain pages and I created a file that works with this type of station. (see picture) 
Mine is a Raddy MN6 lite. 
[image]

This model exists on different brands.
if you using rtl_433 and your driver is Emax - EM3551H it will work. 
it's easy to modify.

Here is the step to decode it with weewx

1) sdr.py doesn’t provide a function that converts degrees Fahrenheit to Celsius so paste this code to sdr.py after
 **# utilities for inline unit conversions.  respect the None!** (utility functions start from line 169)

```
def to_C(v):
    if v is not None:
        v  = 5 / 9 * (v - 32)
    return v
```
    
 2) Always in sdr.py after the **class packets**:    towards the line 393-394 paste this code.
 
```
class EmaxEM3551HPacket(Packet):
    # {"time" : "2024-10-25 17:51:33", "model" : "Emax-EM3551H", "id" : 1001, "channel" : 4, "battery_ok" : 1, "temperature_F" : 55.700,"humidity" : 95, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "wind_dir_deg" : 169, "rain_mm" : 0.000, "mic" : "CHECKSUM"}

    IDENTIFIER = "Emax-EM3551H"

    @staticmethod
    def parse_json(obj):
        pkt = dict()
        pkt['dateTime'] = Packet.parse_time(obj.get('time'))
        pkt['usUnits'] = weewx.METRIC
        sensor_id = obj.get('id') # changes every time you reset the outdoor sensor
        pkt['battery'] = 0 if obj.get('battery') == 'OK' else 1
        pkt['temperature'] = to_C(Packet.get_float(obj, 'temperature_F'))
        pkt['humidity'] = Packet.get_float(obj, 'humidity')
        pkt['wind_gust'] = Packet.get_float(obj, 'wind_max_km_h')
        pkt['wind_speed'] = Packet.get_float(obj, 'wind_avg_km_h')
        pkt['wind_dir'] = Packet.get_float(obj, 'wind_dir_deg')
        pkt['total_rain'] = Packet.get_float(obj, 'rain_mm')
        pkt = Packet.add_identifiers(pkt, sensor_id, EmaxEM3551HPacket.__name__)
        return pkt

```

3) close and save sdr.py

4) The sensor map must be put in weewx.conf 

```
##############################################################################

[SDR]
    # This section is for the software-defined radio driver.

    # The driver to use
    driver = user.sdr
    cmd = rtl_433 -M utc  -F json

    # collect data from Emax-EM3551H
    [[sensor_map]]
        windDir = wind_dir.1001.EmaxEM3551HPacket
        windSpeed = wind_speed.1001.EmaxEM3551HPacket
        windGust = wind_gust.1001.EmaxEM3551HPacket
        outTemp = temperature.1001.EmaxEM3551HPacket
        outHumidity = humidity.1001.EmaxEM3551HPacket
        rain_total = total_rain.1001.EmaxEM3551HPacket

##############################################################################
```

5) close and save weewx.conf 
6) reboot or restart service. 
That's all !!! Enjoy

support which helped me understand.
thanks to the author ivan
https://sh.com.hr/author/ivan/

matthewwall, 2024-12-28:
thank you for this!  added to sdr at commit b1d4441

closed 2024-12-28 by matthewwall