matthewwall/weewx-sdr #193: Requesting support for Ecowitt/Fineoffset air quality sensors

ChrisM510, 2025-02-04:
I would like if support for Ecowitt/Fineoffset air quality sensors could be added.  I bought a WH41 (outdoor) and a WH46 (indoor) sensors, and would like to bring them into WeeWX.

Here is what I was able to pick up using `sudo PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/sdr.py --cmd="rtl_433 -M utc -F json"`

`out: ['{"time" : "2025-02-04 13:10:21", "model" : "Fineoffset-WH0290", "id" : 197, "battery_ok" : 0.800, "pm2_5_ug_m3" : 9, "estimated_pm10_0_ug_m3" : 8, "family" : 65, "unknown1" : 0, "mic" : "CRC"}\n', '{"time" : "2025-02-04 13:10:21", "model" : "Fineoffset-WH0290", "id" : 197, "battery_ok" : 0.800, "pm2_5_ug_m3" : 9, "estimated_pm10_0_ug_m3" : 8, "family" : 65, "unknown1" : 0, "mic" : "CRC"}\n']
out: ['{"time" : "2025-02-04 13:09:57", "model" : "Fineoffset-WH46", "id" : 10514, "battery_ok" : 1.000, "temperature_C" : 22.500, "humidity" : 35, "pm1_ug_m3" : 2.300, "pm2_5_ug_m3" : 2.400, "pm4_ug_m3" : 2.400, "pm10_ug_m3" : 2.400, "co2_ppm" : 541, "unknown" : 400, "ext_power" : 1, "mic" : "CRC"}\n', '{"time" : "2025-02-04 13:09:58", "model" : "Acurite-Tower", "id" : 7642, "channel" : "B", "battery_ok" : 1, "temperature_C" : 9.300, "humidity" : 77, "mic" : "CHECKSUM"}\n', '{"time" : "2025-02-04 13:09:58", "model" : "Acurite-Tower", "id" : 7642, "channel" : "B", "battery_ok" : 1, "temperature_C" : 9.300, "humidity" : 77, "mic" : "CHECKSUM"}\n', '{"time" : "2025-02-04 13:09:58", "model" : "Acurite-Tower", "id" : 7642, "channel" : "B", "battery_ok" : 1, "temperature_C" : 9.300, "humidity" : 77, "mic" : "CHECKSUM"}\n']
`

Would this be possible?  What can I do to help?

Thanks in advance!

andylittle, 2025-02-05:
I assume Fineoffset-WH0290 is the WH41

`{"time" : "2025-02-04 13:10:21", "model" : "Fineoffset-WH0290", "id" : 197, "battery_ok" : 0.800, "pm2_5_ug_m3" : 9, "estimated_pm10_0_ug_m3" : 8, "family" : 65, "unknown1" : 0, "mic" : "CRC"}`

`{"time" : "2025-02-04 13:09:57", "model" : "Fineoffset-WH46", "id" : 10514, "battery_ok" : 1.000, "temperature_C" : 22.500, "humidity" : 35, "pm1_ug_m3" : 2.300, "pm2_5_ug_m3" : 2.400, "pm4_ug_m3" : 2.400, "pm10_ug_m3" : 2.400, "co2_ppm" : 541, "unknown" : 400, "ext_power" : 1, "mic" : "CRC"}`

andylittle, 2025-02-05:
Fineoffset-WH0290, this one should already work. Just add to your sensor map.

ChrisM510, 2025-02-05:
> I assume Fineoffset-WH0290 is the WH41
>
> `{"time" : "2025-02-04 13:10:21", "model" : "Fineoffset-WH0290", "id" : 197, "battery_ok" : 0.800, "pm2_5_ug_m3" : 9, "estimated_pm10_0_ug_m3" : 8, "family" : [...]

Yes, after doing some digging this morning, I found out that the WH0290 is a kit whose sensor is the WH41.  I only purchased the WH41 sensor since I am intending on using this exclusively with WeeWX, and didn't need the base station/display.

andylittle, 2025-02-05:
Yes you can swap the file. Try [this](https://github.com/andylittle/weewx-sdr-1/tree/master/bin/user) untested one, I added the outdoor sensor"

ChrisM510, 2025-02-05:
That worked!

`	Line  60: parsed: {'dateTime': 1738787336, 'usUnits': 16, 'battery.10514.FOWH46Packet': 0, 'co2_atm.10514.FOWH46Packet': 497.0, 'pm1_0_atm.10514.FOWH46Packet': 4.5, 'pm2_5_atm.10514.FOWH46Packet': 4.8, 'pm10_0_atm.10514.FOWH46Packet': 5.0, 'temperature.10514.FOWH46Packet': 23.9, 'humidity.10514.FOWH46Packet': 43.0}
	Line  61: parsed: {'dateTime': 1738787336, 'usUnits': 16, 'battery.10514.FOWH46Packet': 0, 'co2_atm.10514.FOWH46Packet': 497.0, 'pm1_0_atm.10514.FOWH46Packet': 4.5, 'pm2_5_atm.10514.FOWH46Packet': 4.8, 'pm10_0_atm.10514.FOWH46Packet': 5.0, 'temperature.10514.FOWH46Packet': 23.9, 'humidity.10514.FOWH46Packet': 43.0}
	Line 123: parsed: {'dateTime': 1738787396, 'usUnits': 16, 'battery.10514.FOWH46Packet': 0, 'co2_atm.10514.FOWH46Packet': 499.0, 'pm1_0_atm.10514.FOWH46Packet': 3.7, 'pm2_5_atm.10514.FOWH46Packet': 4.0, 'pm10_0_atm.10514.FOWH46Packet': 4.1, 'temperature.10514.FOWH46Packet': 24.0, 'humidity.10514.FOWH46Packet': 44.0}
	Line 144: parsed: {'dateTime': 1738787420, 'usUnits': 16, 'battery.197.FOWH0290Packet': 1, 'pm2_5_atm.197.FOWH0290Packet': 9.0, 'pm10_0_atm.197.FOWH0290Packet': 8.0}
	Line 145: parsed: {'dateTime': 1738787420, 'usUnits': 16, 'battery.197.FOWH0290Packet': 1, 'pm2_5_atm.197.FOWH0290Packet': 9.0, 'pm10_0_atm.197.FOWH0290Packet': 8.0}
	Line 182: parsed: {'dateTime': 1738787456, 'usUnits': 16, 'battery.10514.FOWH46Packet': 0, 'co2_atm.10514.FOWH46Packet': 501.0, 'pm1_0_atm.10514.FOWH46Packet': 3.9, 'pm2_5_atm.10514.FOWH46Packet': 4.1, 'pm10_0_atm.10514.FOWH46Packet': 4.1, 'temperature.10514.FOWH46Packet': 24.0, 'humidity.10514.FOWH46Packet': 43.0}
	Line 183: parsed: {'dateTime': 1738787456, 'usUnits': 16, 'battery.10514.FOWH46Packet': 0, 'co2_atm.10514.FOWH46Packet': 501.0, 'pm1_0_atm.10514.FOWH46Packet': 3.9, 'pm2_5_atm.10514.FOWH46Packet': 4.1, 'pm10_0_atm.10514.FOWH46Packet': 4.1, 'temperature.10514.FOWH46Packet': 24.0, 'humidity.10514.FOWH46Packet': 43.0}
`

ChrisM510, 2025-02-05:
Update:

I added this measurement that was missing:
`pkt['pm4_atm'] = Packet.get_float(obj, 'pm4_ug_m3')`

I followed the same syntax you used.  :-)

ChrisM510, 2025-02-06:
@andylittle  Hi Andy, hey, would you know why this sdr.py file only works when I am running weewx from the command line, i.e. `sudo weewxd /etc/weewx/weewx.conf`?  Data for the WH46 doesn't get logged to the database when weewx is running as a service, but logs fine if I run from command line.  The WH41/WH0290 logs to the database fine either way.  

Thanks again for all your help!  Very much appreciated!

andylittle, 2025-02-06:
can you post your sensor map from weewx config?

ChrisM510, 2025-02-06:
@andylittle   I created new columns in the database for the new fields.  I attached a screenshot from my database showing the columns and highlighted where I had terminated the process from the command line and ran weewx as a service - the new sensor you created dropped out while the service was running, but started picking up data again when I switched back.

  [[sensor_map]]
        outTemp = temperature.00F3.AcuriteAtlasPacket
        outHumidity = humidity.00F3.AcuriteAtlasPacket
        windSpeed = wind_speed.00F3.AcuriteAtlasPacket
        windDir = wind_dir.00F3.AcuriteAtlasPacket
        UV = uv.00F3.AcuriteAtlasPacket
        rain_total = rain_total.00F3.AcuriteAtlasPacket
        radiation = lux.00F3.AcuriteAtlasPacket
        strikes_total = strike_count.00F3.AcuriteAtlasPacket
        lightning_disturber_count = strike_count.00F3.AcuriteAtlasPacket
        #strikes_total = strikes_total.00BC.AcuriteLightningPacket #Birdhouse - Lightning Sensor
        lightning_distance = strike_distance.00F3.AcuriteAtlasPacket
        #lightning_distance = distance.00BC.AcuriteLightningPacket #Birdhouse = Lightning Sensor
        extraTemp1 = temperature.00BC.AcuriteLightningPacket    #Birdhouse - Lightning Sensor
        extraHumid1 = humidity.00BC.AcuriteLightningPacket    #Birdhouse - Lightning Sensor
        extraTemp2 = temperature.1DDA.AcuriteTowerPacketV2    #Backyard Level 
        extraHumid2 = temperature.1DDA.AcuriteTowerPacketV2    #Backyard Level
        extraTemp3 = temperature.3:0.AmbientF007THPacket    #Inside Sensor
        extraHumid3 = humidity.3:0.AmbientF007THPacket    #Inside Sensor
        luminosity = lux.00F3.AcuriteAtlasPacket
        outPM2_5 = pm2_5_atm.197.FOWH0290Packet
        outPM10_0 = pm10_0_atm.197.FOWH0290Packet
        inCO2 = co2_atm.10514.FOWH46Packet
        inPM1_0 = pm1_0_atm.10514.FOWH46Packet
        inPM2_5 = pm2_5_atm.10514.FOWH46Packet
        inPM4 = pm4_atm.10514.FOWH46Packet
        inPM10_0 = pm10_0_atm.10514.FOWH46Packet

[image]

ChrisM510, 2025-02-10:
@andylittle Hey, any thoughts on why I am not pulling sensor data when running weewx as a service, but it works when I run it as a command-line process?  :-)  Again, thanks for all your help, I truly do appreciate it!

andylittle, 2026-02-05:
Different user when ran as service? This question would be better [here](https://groups.google.com/g/weewx-user)

hilman2, 2026-10-21:
WH46 support is published in https://github.com/weewx-packages/weewx-sdr/releases/tag/v0.99.0, including PM4. The WH0290 parser already exists. Fixtures verify decoding; for service-only missing WH46 data, compare the actual service config path, sensor_map and execution user.