matthewwall/weewx-sdr #195: bresser hcoh/voc 868mhz

farmabep, 2025-07-12:
Good morning everyone,
If it's possible, I'd like someone with good will to insert this Bresser Air Quality 868MHz HCHO/VOC sensor into sdr.py so I can view the data in Weewx-sdr. I'll insert it below the log to identify the sensor and then insert the appropriate packets into the SDR field of the Weewx Bresser.conf file.
DEBUG user.sdr: parse_json: unknown model Bresser-HCHOVOC
2025-07-12T13:46:15.578511+02:00 raspberrypi weewxd-bresser7in1[1527]: DEBUG user.sdr: punt unrecognized line '{"time" : "2025-07-12 11:46:12", "model" : "Bresser-HCHOVOC", "id" : 27590, "channel" : 1, "battery_ok" : 1, "hcho_ppb" : 3, "voc_level" : 5, "mic" : "CRC"}#012'

Perhaps these are the strings to insert into SDR?
HCHO = hcho_ppb.27590.Bresser7in1Packet
VOC = voc_level.27590.Bresser7in1Packet
batteryStatus3 = battery.Bresser7in1Packet
Thanks to everyone in advance

farmabep, 2025-07-13:
or this?
HCHO = hcho_ppb.27590.BresserHCHOVOCPacket
        VOC = voc_level.27590.BresserHCHOVOCPacket
        batteryStatus3 = battery.27590.BresserHCHOVOCPacket

andylittle, 2025-07-14:
[This ](New device Bresser HCHO/VOC sensor, PN 7009978 · Issue #2814 · merbanan/rtl_433 https://share.google/yapICMpT1xWFZCxGs)explains the values.

farmabep, 2025-07-14:
Sorry...I messed up a bit...I read the references to inserting rtl 433 you sent me, but I don't know how to insert the instructions into sdr.py or how to create the
parsed files to insert into the [[sensor_map]] field.

farmabep, 2025-07-15:
Good evening Andy....I tried.....I inserted these instructions in sdr.py but I couldn't do it

class BresserHCHOVOCPacket(Packet):
    # {"time" : "2025-07-13 10:22:25", "model" : " Bresser-HCHOVOC ",
    # "id" : 27590, "battery_ok" : 1, "hcho_ppb" : 1,
    # "voc_level" : 5,  ""mic" : "CRC"

    IDENTIFIER = "Bresser-HCHOVOC "

    @staticmethod
    def parse_json(obj):
        sensor_id = obj.get('id')
        pkt = dict()
        pkt['dateTime'] = Packet.parse_time(obj.get('time'))
        pkt['usUnits'] = weewx.METRICWX
        pkt['hcho'] = Packet.get_float(obj, 'hcho_ppb')
        pkt['voc'] = Packet.get_float(obj, ' voc_level ')
        pkt['battery'] = Packet.get_battery(obj)
        pkt = Packet.add_identifiers(pkt, sensor_id, BresserHCHOVOCPacket.__name__)
        return pkt

I also tried to insert these 3 lines into the sensor map but...nothing to do

HCHO = hcho_ppb.27590.BresserHCHOVOCPacket 
      VOC = voc_level.27590.BresserHCHOVOCPacket 
    batteryStatus3 = battery.27590.BresserHCHOVOCPacket